Matthew-Hsu/PiPass #1: Wrong interface names

julienbiral, 2015-03-29:
On my PiPass, my wifi adapter and ethernet are shown as wlan1 and eth1 instead of wlan0 and eth0. I am very new to the RPi, but it seems that you gave us the image disk you worked on instead of a fresh one that was set up properly. I am not sure about that but from what I've seen, it makes sense.

Additionally, you shouldn't set a default MAC address as it won't work out of the box for the users of PiPass. Instead, I'd rather put an empty mac file with some comments explaining that the user should put the MAC address of his 3DS and why.

Matthew-Hsu, 2015-03-29:
Thanks for the comments, I really appreciate them.

I put some directions about the MAC address in the readme near the beginning, but yes, I agree that leaving the file empty would be better.

As for wlan1 and eth1, I created those images from a fresh install. The file should be referring to eth0 as br0 and wlan1 and wlan0. The image should match the network file that is listed in the source.

I will look into this and make some changes. Thanks for letting me know.

julienbiral, 2015-03-30:
That's a great project so I am glad to help you improving it by giving a feedback!

I was probably unclear, but by default, the image was having the eth0 and wlan0 referred in the files. However, both my ethernet and wifi dongle were recognized as wlan1 and eth1, so it wasn't working out of the box without changing the names in the hostapd files and interface.

What I suspect is that you were setting up the image disk by using your ethernet and adapter. Then, mine were recognized as a second ethernet and wifi dongle. To be honest, I am not sure at 100% that this is the cause of the issue I've encountered. But it might be, as I don't have any problems with other image disk.

Matthew-Hsu, 2015-03-30:
Thanks! I understand where you are having issues now. I've made the changes to mac_accept, so it is a bit more clear. I've pushed other additional features as well and will look into the interface issue.

It makes sense that the issues are coming from this because if you are not using wlan0, the bridge should not work. In this case, you had to manually edit some files to get it working. I'll look into the interface configuration if I can easily specify some type of dual configuration or automatic detection for this because I will need the active interfaces for piPass.py.

You'll probably also have to modify piPass.py and change wlan0 to wlan1 if you want to have cycles working correctly.

I always thought the Raspberry Pi would pull eth0 since it had only one port.

Thanks again for the feedback and I hope the project is useful for you!

ddcruver, 2015-04-12:
Your problem is that udev will detect new network devices at boot time and give them names based on their mac addresses.  When you boot the image on a different Pi with a new mac address for both the wired and wireless interfaces it will give them new names.

udev will write this information in /etc/udev/rules.d/70-persistent-net.rules.

You can either delete this file or if you want to be safe then just comment out the lines that correspond to your wired and wireless adapters.  Upon reboot it should regenerate new entries for any new network devices.

Additionally you could write some sysprep like script that will delete this file then shutdown the system for when your preparing images or for users to fix their machine when they migrate from one Pi to another.

Matthew-Hsu, 2015-04-19:
Thanks for the feedback, ddcruver. I'll prepare the next set of images this way to avoid this issue. I have some more updates to push, so I'll roll it in probably within the week.

Matthew-Hsu, 2015-04-19:
Removed /etc/udev/rules.d/70-persistent-net.rules in the 1.2 image. Tested image file and /etc/udev/rules.d/70-persistent-net.rules is generated again upon a fresh boot.

closed 2015-04-19 by Matthew-Hsu