Matthew-Hsu/PiPass #3: StreetPass update

michealo, 2015-04-19:
Is this still working for anyone after the streetpass update a couple days ago?

I tried recreating the network connection on my 3DS and recycled the pipass to no avail

Matthew-Hsu, 2015-04-19:
I'm planning to update PiPass to 1.2 this week with some feature updates and I can look into this issue more in depth. With all the updates to Mii Plaza this week, PiPass is still working for me. There was an issue that someone had with a different WiFi adapter and I rolled out a hotfix update to piPass.py.

If you are using the pre-built images or one of the releases from the release branch, you can overwrite /opt/PiPass/piPass.py with the master branch's piPass.py. The main page has some more information regarding manual updates and also make sure your 3DS's MAC address is correctly saved in /etc/hostapd/mac_accept.

If that still does not work for you, could you tell me which WiFi adapter you are using? And, if you stop PiPass and start it up again (no refresh), do you get an initial hit with StreetPasses?

Thanks.

michealo, 2015-04-19:
Thanks for the response.

I checked the /opt/PiPass/piPass.py and /etc/hostapd/mac_accept files, both appear ok.
Using Ralink Technology, Corp. RT5370 Wireless Adapter from the CanaKit bundle.

I can connect to the PiPass AP and verify internet, but no new streetpasses have come in since updating mii plaza.  Recycling the Pi didn't help but I'll try to disable streetpass on my n3ds and enable again to see if it it helps.

michealo, 2015-04-19:
Disabling and enabling street pass seems to have fixed my issue, little grew light came on again. 

Thanks, really enjoying PiPass!

closed 2015-04-19 by michealo

Matthew-Hsu, 2015-04-19:
Thanks for the update on how you got it working again. I'll put your solution on a troubleshooting page in the future.

Glad you are enjoying it! Thanks again!