Matthew-Hsu/PiPass #10: [Question] Does Pipass have auto-cycling mac addresses?

Corruptinglyneedful, 2015-04-29:
Currently using Nzone and sick of it. ,needs constant baby-sitting.

Should not be a how a program operates.

Pipass seems to be this program.

Just one problem, main page says that macs have to be added in?  *confused*

The whole process after installation is what is confusing me.

What is root access?  I do know that sudo is the command line for Linux.

I know extremely little about Linux, these commands confuse me.  The whole tutorial confuses me except for "the image is pre-loaded for your SD card.".  I don't get or understand the terms and commands as someone entirely new to Linux or Raspberry Pi.

Is this not a program where you load it, it cycles, and apart from user customization, just sits there and does it's thing all day?, like Nzone?  Nzone is the only Homepass Homebrew software I have experience with and it hasn't been pleasant.  

That's why I'm really looking for, a program that DOES NOT need constant babysitting.

nagledb, 2015-04-29:
There are two kinds of mac addresses involved.

The first kind of mac address is the kind used by Pi when it's spoofing a Nintendo Zone. You do not have to enter those. They're retrieved automatically from a spreadsheet.

The second kind of mac address it the one for your Nintendo 3DS. Since the Nintendo Zones do not use password based authentication, all devices are blocked from accessing it unless their mac address is listed in the file /etc/hostapd/mac_accept. This is the part you have to manually take care of yourself.

You are correct that this is intended to sit there and do its thing all day once you have it up and running.

Regarding root access: on Linux machines, "[root](http://tldp.org/LDP/lame/LAME/linux-admin-made-easy/root-account.html)" is a special user account that has privileges to do just about anything it wants. Using "sudo" temporarily gives you the same level of privilege as root. Logging in as root gives you those privileges for the whole time you are logged in. The root account is also sometimes referred to as the [superuser](https://en.wikipedia.org/wiki/Superuser#Unix_and_Unix-like) account.

In terms of what hardware to get if you decide to try PiPass, I got the "CanaKit Raspberry Pi 2" that's recommended under "Hardware" in the instructions. I used the 8 GB MicroSD card it came with and it works just fine. The kit also comes with a wifi dongle and power supply, both of which you'll need, as well as a case for the pi. It's probably the easiest way to get started for this project, especially since you're new to Raspberry Pi.

Corruptinglyneedful, 2015-04-30:
So I have to add the Mac Address for every 3DS I want to use for it, including friends that come over?

I ordered the Pi 2 kit from the link.

Realistically for someone entirely new and dumb to LInux, how hard and how long does this take to setup?  I also have a switch and ethernet cables so it can continue to get updates or w/e.

nagledb, 2015-04-30:
Correct, you will have to add the Mac Address for every 3DS you want to have access, including visitors. (If this wasn't the case, then your neighborhood would be able to use your internet connection freely since the connection is otherwise unsecured.)

I've been using Linux for a long time so I'm not really sure how easy/hard it would be for someone who's completely new to it. Sorry I can't give a more helpful answer to that. :) The premade image keeps the process relatively simple so I think it should be doable. The instructions provide links to additional resources on the Raspberry Pi website that may help you figure things out.

Corruptinglyneedful, 2015-04-30:
As long as I have a tutorial that explicitly and painfully explains everything step by step and is written for users who know absolutely nothing about Raspberry Pi and Linux, I should do just fine.

Honestly, from going through it, most of the work seems done except mac address configs.

So do you have to log into PiPass?  Is it not a program but an OS?  Is it like Windows login?

Matthew-Hsu, 2015-04-30:
PiPass is just a program that is controlled through a graphical web page to make things easier. I usually have my phone on me, so I wanted something easy to control from my phone, instead of going to my computer each time. Once everything is setup, you just enter the IP address of your Raspberry Pi into a web browser and away you go.

It's not a OS, but just a program that runs on top of Linux. The pre-made images itself is really just a clone of my setup, so all you need to do is just copy that over onto your Pi.

As for the MAC address additions, yes, I do want to make that part easier, so you really won't need to touch much Linux at all. I could have disabled security by default, but it is something everyone should do, especially if you are running it for a while.

Corruptinglyneedful, 2015-04-30:
Nzone doesn't have security, it's just open.  So it's nice that this will have security.  I really don't have anything against Linux, I've used Gparted before, which is based on Linux it's just I'm new to the command line of Linux.

OK so Pipass is not an OS, got it.

"Once everything is setup."  What do you mean here?  I thought that you set up the pi, download the SD image, apply it, and it's done?  Then after that you find the IP, log into the console, add your nintendo 3ds mac address, change your timer settings, and you're done.

Then it just sits and does it's thing, all day long.

Am I wrong?  Did I miss a step?  Because if I missed a step, then I need to know because the Pi 2 comes today.

nagledb, 2015-04-30:
At a glance, your 8-step list looks correct to me. :)

PiPass currently cycles through 59 mac addresses by default. You can view them on tab 1 of this spreadsheet: https://docs.google.com/spreadsheets/d/1OfgyryUHeCPth76ziFT985XNLS-O5EXtjQDa0kA1L6M/edit#gid=85967306

You can also configure it to use a different tab of that spreadsheet, or can even set up your own spreadsheet if you want.

Corruptinglyneedful, 2015-04-30:
Can it be configured to use all of those tabs at once?

I'm guessing that you want to stick with what everyone else is doing as the system is based on community use of Mac addresses.

nagledb, 2015-04-30:
You can't configure it to use multiple tabs, no.

However, you can create your own Google spreadsheet and copy the data from all those tables into a single tab on your own spreadsheet. Then you can configure your PiPass to use that tab on your own spreadsheet. The process is described under [PiPass Customization](https://github.com/Matthew-Hsu/PiPass#pipass-customization).

Corruptinglyneedful, 2015-04-30:
Pipass is not launching and not saving settings.

I got the Pi setup correctly, booted and I can see the correctly named SSID's on wifi devices and 3DS.

Not getting passes.

Requesting immediate assistance.

Corruptinglyneedful, 2015-04-30:
OK, I got Pipass to launch.  The dashboard says that it is currently running and tells me the relay.

When I go to settings and then I say 5 minutes and Hit save, it says to refresh.

I hit refresh Pipass and it says config updated, but when I go back to the config, it says 15 minutes.

3DS sees attwifi in connections, do I need to actually connect to this or no?  Tutorial does not specify.

Still have no passes.

Corruptinglyneedful, 2015-04-30:
Why will this not save my setting of 5 minutes!

I'm getting quite frustrated.

I just want passes every 5 minutes, not 15.

How come there is no text output from PiPass on the Pi's display?  Do I need to re-log into the Pipass on the Pi every single time I reboot the Pi?

nagledb, 2015-04-30:
You won't see text output from PiPass on the display. The process for it runs in the background.

Once you have it working, you shouldn't need to log back into the pi each time you reboot it. It should just work. At most, you might have to use the dashboard to start/restart the process. (As an aside, you shouldn't need to reboot the pi. You can probably just leave it running all the time.)

I think there's a browser caching issue when you make changes that makes it seem like they didn't apply even though they did. Try to do a force refresh on the page after you make your settings change.(For example, in Chrome you can do a force refresh with CTRL+F5.)

To see if it's cycling, watch the main dashboard page. Note what MAC Address it has listed. Then check back 5+ minutes later (since you are using a 5 minute interval) and see if the MAC Address has changed.

No, you do not need to configure your 3DS to use attwifi as a wireless connection. In fact, some of the mac addresses don't even use that as their SSID. Just like any SpotPass location, it should do the StreetPass relay automatically without you having to configure your 3DS.

Did you enter your 3DS mac address into /etc/hostapd/mac_accept ? Maybe double check that for typos. If you have a typo in the mac address, it'll prevent your 3DS from connecting even if you have everything else working properly.

Corruptinglyneedful, 2015-04-30:
1. I originally used - instead of : in the mac address log.  It has been corrected.

2. Pi is booted and logged in, so not display output.  So what is this entered forwarding state text am I getting?

3. OK, don't connect to network attwifi

4. I really hope I do not have to start and stop the Pipass process all the time.  The reason I ditched Nzone was because of constant baby-sitting.  

5. You are correct, there was indeed a caching issue.  I hit Ctrl+F5 in Firefox and it now shows 5 minutes.

6. I think it may be cycling now.

7. Yes, it's cycling.  At least on the PiPass dashboard.

8. Why can I not get passes when my 3DS is open?

nagledb, 2015-04-30:
2 - I haven't encountered an "entered forwarding state" message, I'm not sure what you're referring to.

4 - Currently there's an issue under discussion at #4 where sometimes the cycling stops. @Matthew-Hsu thinks he found the issue and may have a fix released soon. In the mean time, you might have to visit the web dashboard occasionally to restart the process. I generally have to do this about once a day. Once he has that fix made, it should run indefinitely.

8 - StreetPass only works when your 3DS is closed. That's simply how the StreetPass feature works. If you and a friend each have a 3DS and you bring them within range, they also won't StreetPass if one if open. Both have to be closed (but still powered on) for StreetPass to work. You'd have to ask Nintendo why they designed it that way. :)

9 - I think the 8 hour timer only applies when you actually StreetPass on a mac address. So if you don't StreetPass, I don't think it'll start the timer. I'm not sure if it StreetPasses though if your StreetPass apps are all "full" (ie. if your Plaza has 10 people in it, and all your other apps have their max passes as well) so I'm not sure if that starts the timer or not.

Corruptinglyneedful, 2015-04-30:
2. It's some message the Pi is giving me, over and over. with different number things, can't really explain and it's a long string of text.

4. I see, that's not so bad and looking forward to that fix

8. That's not entirely correct according to Nintendo, you're supposed to get passes when playing games.  Reddit /r/3DS has also told me the same thing, I'm just not getting them when anytime I leave it open.

9. So, if your games aren't full and you get a pass, does that count?  Or is it only on Mi Plaza passes?

10. It is now cycling :) and it's already 10x better than Nzone.

I'm quite proud of myself.  I went in knowing next to nothing and got it all set up properly, just with a few hitches due to cache issues.

nagledb, 2015-04-30:
Congrats on getting it up and running! :)

3 - I didn't know that. I don't think I've ever received a streetpass when my 3DS was open, even from another 3DS.

4 - I honestly don't know if it counts or not.

Matthew-Hsu, 2015-05-01:
Closing issue since you have it up and running now! Congrats!

PiPass 1.3 should be out soon this week. It features some nice additions from @nagledb and should have a working fix that prevents PiPass from stopping prematurely.

The changes are up on the master branch and if you feel like upgrading "unofficially" to check things out for now, you can delete /opt/PiPass and /var/www/ on your Raspberry Pi and replace them with ones on the master branch. It's a bit more involved doing a manual upgrade, but I should have an updated image by Sunday.

closed 2015-05-01 by Matthew-Hsu

Corruptinglyneedful, 2015-05-01:
How do you actually update it?  Tell me it's easy.