Matthew-Hsu/PiPass #9: Logging

nagledb, 2015-04-29:
I think it'd be helpful if piPass logged its activity to a log file. Perhaps to /var/log/pipass? This log could include the kind of output messages we get currently on STDOUT when running it manually, plus could contain some additional messages for debugging if needed.

An example: I'm wondering if #4 might be occurring due to problems opening one of the files it reads/writes. If we wrapped the io.open blocks with a try block, we could catch any exception it raises, note what happened in the log, then gracefully quit. That would give us post mortem info that would help us determine if that was somehow the culprit.

I'd be happy to take a stab at doing this using Python's [logging module](https://docs.python.org/3.4/library/logging.html) if you want, but since it's such an intrusive change I wanted to discuss it first.

Matthew-Hsu, 2015-04-30:
I agree, logging would be extremely helpful. I was able to find some time tonight and make some changes locally on my end that will hopefully fix that crashing bug. Depending on the results, I will push something either tomorrow night or the weekend.

You've contributed some pretty great features and I really appreciate it. A logging system that can be turned on or off would be great to have, but I am not sure if we need to implement something that exhaustive yet. Once that crashing bug is fixed, PiPass should be relatively solid.

PiPass is really a fun project for me to try some things out that I haven't really tried yet. Python logging seems interesting, so if it is something you would enjoy doing, feel free to give it a shot. I may eventually get around to it sometime, but don't feel compelled to do so.

Again, thanks for all your contributions so far and I really enjoy the features you've added!

nagledb, 2015-04-30:
I'm not sure why this didn't occur to me sooner but there's a much easier solution for logging: just redirect STDOUT and STDERR to a log file when piPassCommand starts up piPass. Pair that with a conf file in /etc/logrotate.d to ensure it doesn't grow out of control and that should be more than sufficient.

It'd probably be pretty simple to implement, but given the issues/changes under discussion in #4 I'll hold off on it until we see what comes of that since it may change where/how piPass gets started.

Matthew-Hsu, 2015-05-01:
I've pushed the changes on the master branch already. Issue #4 should be fixed now. piPassCommand has been removed. You can still get output manually from piPass, but I'm thinking of just outputting those regular messages to a file instead and just performing a cleanup where applicable.

Matthew-Hsu, 2015-05-17:
I've implemented a logging system in PiPass using the Python logging module. It works quite well and it was pretty smooth to implement.

Since, the command console was semi-retired, I've brought it back in another form through PiPass Logging in the dashboard.

closed 2015-05-17 by Matthew-Hsu