Matthew-Hsu/PiPass #12: Link to Update Option Missing

pokeguru87, 2015-05-02:
This is a cosmetic issue, no functionality is broken. The option in the menu to "Update" is missing on 2 pages. The "Dashboard" and PiPass "Stop" pages. All other pages have the link available.

nagledb, 2015-05-02:
The link for Update appears for me on both of those pages. Maybe your browser is showing you a cached older version?

pokeguru87, 2015-05-02:
Ya, that seems to be the case. Odd, that none of the other pages I used more often didn't show up, maybe due to them being the setting pages so the forced update stuff he added prevented it. Still, my bad.

closed 2015-05-02 by pokeguru87

Corruptinglyneedful, 2015-05-03:
This happened to me too, when I first booted the Dashboard, than it just later reappeared.  I even did a force reload.

Not sure what was up with that.