Matthew-Hsu/PiPass #15: Configuration File

Exor, 2015-05-03:
Now that we can update PiPass from the web interface, I would like to suggest adding a configuration file with paths. As it stands now, every time PiPass is updated, I need to go into the pipass.py file and update my folder paths. The update script also deletes the entire www directory structure. instead of just the path it is installed in. 

I use Nginx instead of Apache, so my www directory is at /usr/share/nginx/www rather than /var/www. I also use my Pi as a security camera with the root index.html pointing to my camera feed. I have pipass in its own folder under www. It would be awesome if PiPass kept track of the folders where it resides and doesn't touch other files.

Corruptinglyneedful, 2015-05-03:
I was under the impression you just clicked update and it worked?

nagledb, 2015-05-03:
I wonder if it would be well addressed by moving the configuration files to live under /etc/pipass/ rather than storing them under the web directory? That would also better support someone who might want to run piPass.py without using the web interface at all.

As a work-around for now, maybe you could use symlinks? If you want your PiPass at /usr/share/nginx/www/pipass, then maybe you could do:

    ln -s /var/www /usr/share/nginx/www/pipass

That way the code lives where PiPass expects it to, but nginx could still pick it up where it expects to find it.

Matthew-Hsu, 2015-05-03:
@Corruptinglyneedful You are correct. Update works as you think it does for the pre-made images or if you install PiPass the same way if you go the manual route.

@Exor I'll look into it and make it more customizable in the future. Thanks for the feedback!

Matthew-Hsu, 2015-05-09:
@Exor With your current configuration, I don't think you will be able to use the update function to get the newest changes. However, moving forward, PiPass is now configurable to use a different web service and the updater has been improved to only clear files from the target path specified. Since you installed PiPass (e.g., /usr/share/nginx/www/pipass) somewhere else, the updater should only clear files in that target directory (e.g., pipass). PiPass settings are automatically backed up, so these changes should help you out without changing code manually when switching versions. If this addresses your issue, please let me know so that I can close this issue off. Thanks!

@nagledb The PiPass Dashboard is a requirement for the time being since the GUI needs access to the configuration files on the web service. JavaScript cannot access system files outside of a web directory. A workaround can definitely be done in the future to provide a separation between GUI and terminal, but with the updater in place, if these changes were to occur, configuration files would exist in /opt/PiPass/config/ instead.

Exor, 2015-05-11:
The new version works great! Thanks for the update.

closed 2015-05-11 by Exor

Matthew-Hsu, 2015-05-12:
Glad it worked out for you! Enjoy!