Matthew-Hsu/PiPass #17: Does PiPass have it's own Reddit?

Corruptinglyneedful, 2015-05-09:
I can't seem to find one.

Pipass relies on more people using it, right?  The bigger it is, the more varied passes and users you see?

I think building a community around it, is a good idea.  It should help generate more traffic and ultimately different passes.  I realize that there will be questions, much like here of how to set it up and what not but it's more or less the same.

Idk.

@Matthew-Hsu, I ran into your Mii.  You the real VIP and as such have a place in my VIP room.

Matthew-Hsu, 2015-05-09:
PiPass doesn't have its own sub-reddit yet. I just have some threads on Reddit, but so far, the GitHub has been the central place for communication.

I do agree, a bigger community would be better. The passes you are getting though are not exclusive to PiPass. Since I am using the community list of MACs, we're all combined with people of different methods. This would include NZone and other solutions.

I might have StreetPassed you as well! But, yes, if you want to help, I would appreciate it. Before starting this project, I came in with zero knowledge about HomePass. I never used NZone or any other solution. I read how it worked and decided to build one myself. I liked it because I really didn't know what others did before making PiPass. I felt like having a side personal project to see what the Raspberry Pi was all about was a good idea, so I do understand and don't mind if people have questions. I can't guarantee immediate responses, but it looks like we're building a mini community to help out. I do try to at least provide a response to everyone though.

If you have any more ideas, I would like to hear them. A lot of the feedback so far has shaped PiPass into something that, I believe, does Homepass in a more flexible, not constrained, and a transparent way than other solutions out there.

Corruptinglyneedful, 2015-05-09:
@Matthew-Hsu  Oh you definitely did street pass me.  Look out for the cowboy and the personal message that I sent you, that says Corruptingly, as that was a clue to my username here.

>A lot of the feedback so far has shaped PiPass into something that, I believe, does Homepass in a more flexible, not constrained, and a transparent way than other solutions out there.

ABSOLUTELY agree.  Since the mac bug has been fixed, I have not had to restart it, sans for a power outage.  Pipass is absolutely one of the superior methods of Homepass.  It requires very little effort, hardware, and I think that following my thread I created along with the baseline tutorial, anyone can correctly set it up.

I'm a completely newb when it comes to anything outside of Windows and I got it working.

I couldn't get it to update manually though and I probably would not have luck with manual installation, but flashing the image, running a few commands, etc, that I can do.

I have gone from 200 passes to nearly 900 in less than 2 weeks.  Now that I've slowed it down, (started playing The Witcher series), I'm averaging about 40 per day or two rounds.

I must admit, I was skeptical at first but hopeful.  Even my friends were like, you're going to undertake a project, buy hardware, etc for this thing you aren't sure works and have no clue what you are doing?

Yup.

I am very satisfied with the project you have created here.

In regards to your mention about the public addresses, do any come from OFFICIAL nintendo relays?

Matthew-Hsu, 2015-05-09:
I shall keep a look out for you then!

I'm really glad you are enjoying PiPass and it's pretty awesome that you learned quite a bit from all of this. It was a pretty bold move. I know PiPass is relatively new and didn't have much news in its early stages, so the fact that you went out and gathered all the material and got it working is pretty impressive. 

Another feature I want to eventually implement is to offer a interface to add/remove the accepted 3DS MAC addresses without the need to manually use terminal to do it. After that change, I do believe pretty much everything in PiPass can be configured through the dashboard.

You hit the StreetPass pretty hard it seems! Which is good, we needed the stress test on PiPass!

To answer your question, yes, the default list used are actual official Nintendo Zones. So, this would be a mixture of "real" people alongside with "homepass" people. If you have switched to use the extended MAC list, I do believe that list is exclusively for "homepass" people.

Thanks for using the referral link! I really appreciate it :)

Corruptinglyneedful, 2015-05-09:
Your referral link was posted in the tutorial and I just went through, clicked it, bought the hardware because you said it worked.  I didn't want to run the risk of buying something and it not work.

I would definitely appreciate a way to authorize 3DS's through the Dashboard, not having to ever connect to the Pi beyond initial setup is ideal.

Plus I only have one keyboard..

I do not mind Real people from real relays, it should be more varied then, as if you're only using Homepass people, you are liable to keep getting the same people over and over again.

Extended Mac list?

Matthew-Hsu, 2015-05-09:
Hmm... not sure if you played around with this setting yet, but I've linked this spreadsheet in the tutorial: https://docs.google.com/spreadsheets/d/1OfgyryUHeCPth76ziFT985XNLS-O5EXtjQDa0kA1L6M/pubhtml

If you look at this, you'll see that there are numbers followed by a worksheet name. If you go to PiPass Settings and enter the corresponding number into PiPass DB Worksheet, you'll tell PiPass to use that list instead. Extended MACs is numbered 6 and is a much larger list than the default list being used. You can read more about it under Customizing PiPass in the tutorial

Corruptinglyneedful, 2015-05-09:
Can't you just combine all the lists together into a Mega giant list so you get even more variation in passes?

Matthew-Hsu, 2015-05-09:
I could, but I'm following the standard as set by www.homepass.info and there are some lists that should not be mixed, since these are special zones that are targeted. For example, only use this one zone if you have a Red Mii.

The tutorial shows you how to add a custom worksheet where you can edit and use personally, but I guess I could mix worksheet 2 and worksheet 6 together and add them to worksheet 1, if you don't want to play around with settings and want a varied approach.

Matthew-Hsu, 2015-05-09:
Many more zones have been added now. If you use worksheet number 7 (PiPass DB Worksheet = 7), you'll get a list containing both Nintendo Zones and Extended MACs. There are 701 zones in this list, so it is quite plenty.

I'll close this issue, but feel free to talk in here if you want and I can follow up too.

closed 2015-05-09 by Matthew-Hsu

Corruptinglyneedful, 2015-05-09:
700 addresses?

Holy cow.

So that's both official relay zones and extended Homepass people?

Corruptinglyneedful, 2015-05-09:
What does this mean?

IMPORTANT: It has been found that these MAC's will not interface with "real" people in those locations unless we get & set the Access Point SSID to the same. Just using "attwifi" will not work under these circumstances. The SSID must be the same if you want to StreetPass Relay with actual people in those locations. If you just want to HomePass with people over the internet, then you can continue using "attwifi" - otherwise we need to harvest the AP SSID as well.

Matthew-Hsu, 2015-05-09:
That's the official and extended lists, yes. I missed a bunch from the extended list on www.homepass.info, so I've added them in here.

The way Nintendo has set things up with official Nintendo Zones is that it is not only good enough to get the MAC address of that zone, but to also get the correct SSID name as well. If both are correct, then you will actually get people who walk by those zones. If this information is incorrect, then you'll only get people who Homepass through those zones.

Either way, you'll be getting people regardless. It just depends if you want people from certain areas, then the Nintendo Zone list is helpful.

Corruptinglyneedful, 2015-05-10:
So is it correct then or is it something I have to do?

I need to know and so should users if we need to change things.  I want both real and Homepass people.

Matthew-Hsu, 2015-05-10:
It's correct. The community lists always try for "real" people before using "homepass" people. If you want a mixture, use worksheet 7.

In the end, you're getting actual people and there isn't much we can do at this point.

Corruptinglyneedful, 2015-05-10:
That's good though, more variation and more BIRTHDAYS.

I have nearly 260.

Matthew-Hsu, 2015-05-10:
260!? Haha, very nice!

Corruptinglyneedful, 2015-05-10:
What's even nicer is that I bought the premium upgrade right about the same time I started using Nzone on Windows, which was such a pain, it caused me to switch to PiPass.

So, I got 260ish bdays in less than 1 month.