Matthew-Hsu/PiPass #18: [Minor issue] PiPass DB extended prime MAC incomplete (page 6)

LoganMcClay, 2015-05-09:
Hi Matthew !

I just want to notify you the extended prime MAC list is incomplete, if you look at FatMagic's list, the ones beginning by 02:53... 6E:53... and 40:53... should not be only ending by 46 but use the full list from 00 to 9F.
I extended my custom list accordingly here, if you want to update yours :
https://docs.google.com/spreadsheets/d/1yFOUya1wgMiBkYk1PcdR4vDwhZ8xSKxo7Ps7EbgmWfs/edit

Matthew-Hsu, 2015-05-09:
Hello Logan, thanks for the feedback! I do admit that the default list that PiPass uses is more of a proof of concept than anything else. The goal is to just show others what an ideal format should look like in case if the community ever wanted to parse from a "master" list. I do have a custom tab in the worksheet, so another idea behind this decision was to encourage others to have their own list for any custom control. I do understand though that there are many who just prefer to have a standard default list.

If the list is problematic, I will go ahead and make some changes. Though, I will look into this and since there is quite a few additions since the last time I built the list, I can go ahead and make the amendments to the default list.

Thank you for sharing your document with me!

closed 2015-05-09 by Matthew-Hsu

Matthew-Hsu, 2015-05-09:
I've went ahead and updated PiPass DB with your added values. There seems to be some new additions for the Nintendo Zone list as well on www.homepass.info, so I went ahead and updated that list as well. Thanks again!

LoganMcClay, 2015-05-09:
You're welcome ! And by the way, thanks for your PiPass :+1: