Matthew-Hsu/PiPass #20: About 8188CU

grimlokason, 2015-05-18:
Hello, 

Sorry, i'm quiet a noob on github so i create an issue for this information ^^'...

I've a RTL8188CU on my rpi for homepassing.
The tips is to use the drivers from the realtek website (and not the one in the kernel) and to find an hostapd also on their website.

Information from my rpi : 

rpidomo ~ # hostapd -v
hostapd v0.8.x_rtw_r7475.20130812_beta
User space daemon for IEEE 802.11 AP management,
IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator
Copyright (c) 2002-2011, Jouni Malinen <j@w1.fi> and contributors

rpidomo ~ # modinfo /lib/modules/3.6.11+/kernel/drivers/net/wireless/rtl8192cu/8192cu.ko
filename:       /lib/modules/3.6.11+/kernel/drivers/net/wireless/rtl8192cu/8192cu.ko
version:        v3.4.4_4749.20121105
author:         Realtek Semiconductor Corp.
description:    Realtek Wireless Lan Driver
license:        GPL
srcversion:     5E5C45B85858009E4ABA963
[...]

If you want to give a try, i've a spare rpi with a wifi usb dongle that i can use for testing images. 

Regards.

Matthew-Hsu, 2015-05-18:
Hello,

Thanks for the information. Yes, I was unable to try the RTL8188CU myself as I don't own one. Just to confirm, you have the RTL8188CU working with Homepass? There is a lot of information about the RTL8188CU and I've heard that it is unable to change MAC addresses.

I will look into pre-installing the RTL8188CU drivers from the Realtek website and building a version of Hostapd that supports it too. I don't have a timeline for this yet, but if you want to give that a test, that would be much appreciated.

Another quick question for you, where on the Realtek website did you find Hostapd?

Thanks!

grimlokason, 2015-05-19:
Hello,

Yes, i can homepass with it.

In fact you can spoof the MAC addresses only with the RTL drivers and this drivers work only with their hostapd (it drive me mad for some night before i found that ^^').

For the Hostapd you should find it in the driver package here : http://www.realtek.com.tw/downloads/downloadsView.aspx?Langid=1&PNid=48&PFid=48&Level=5&Conn=4&DownTypeID=3&GetDown=false&Downloads=true#2742

Matthew-Hsu, 2015-05-19:
This is great to hear! I'm sure this will help many others. 

I will give it a go and make sure their hostapd works with other WiFi dongles too. If all goes well, I may have an image for you to try out.

Thanks again for the information. It is greatly appreciated!

listix, 2015-05-19:
I am having trouble with the homepassing with a RTL8188CU.
What did you exactly do with the hostapd from realtek?

grimlokason, 2015-05-19:
@Matthew-Hsu : I don't think it'll work with other dongle :/ (but i just have 2 RTL8188...)
It'll need to been installed with a different prefix. (and option on web page spécific for the RTL dongle)

@listix You should compile and install it. the way to use it is the same as the official hostapd.

listix, 2015-05-19:
I had to go through so many problems to be able to compile the driver and the hostapd. I now think I messed the hostapd.conf during so many tests so I will have to fix that. Did you have problems when compiling it? What kernel version do you have?

I will do the compiling again but with the image provided in the main page to see if that makes a difference.

grimlokason, 2015-05-19:
@listix It was a year ago so i don't remember well about compilation. i'll post my hostapd.conf when i'll be back home.

My kernel is a 3.6.11+ and I don't remember if he was from the gentoo repo or the official rpi one.

grimlokason, 2015-05-19:
@listix 
Lunching the hostapd with : hostapd /tmp/tmp.PBMqVRGqDc/nztmp -B -d
and the file /tmp/tmp.PBMqVRGqDc/nztmp contain : 

SSID: NZ@McD1
Country: FR
interface=wlan0
channel=1
ssid=NZ@McD1
auth_algs=1
wpa=0
country_code=FR

Regards

Matthew-Hsu, 2015-05-20:
@grimlokason 

I tried giving it a go and I'm using too new kernels to compile the driver. I have tried using the compatible linux headers required to build the driver, but I won't be able to insert the driver into my installation. I'm guessing if I were to use an image with an older kernel, it could possibly work.

With some quick reading, it seems that we can actually install the official drivers through the package firmware-realtek. It's a hunch, but if I could track down a compatible hostapd file, it could possibly work assuming the driver and hostapd are the same as what you are using.

listix, 2015-05-20:
I think I was able to force the use of the driver I compiled. Also I uninstalled the old hostapd and installed the one I compiled. I was able to detect the network attwifi with my laptop but couldn't connect to it. I realized it was because I never added my laptop's mac address to the list but with the old hostapd it allowed me to connect even though the mac address wasn't added. That leads me to believe that the realtek hostapd at least works better with this chipset. But I couldn't make the street pass work yet. At this point I am not sure if the specific hardware of my wireless adapter is the issue or some configuration  is guilty of this.

schneekluth, 2015-05-25:
I have been fightening all day getting __PiPass__ with a __Realtek rtl8188cus__ wifi dongle to work on my old __RaspberryPi Rev.B__. There are way too many tutorials out there stealing your time. Finally it's all running smoothly. Therefore a quick guide for the clueless starting from a fresh raspbian:

01. Install Dependencies (apache, php5. etc.)
02. Clone PiPass git repo and cd into 
03. Copy files from PiPass to it's right places (/etc, /var, /opt)
04. Setting Permissions for user www-data with "sudo visudo" as explained above
04. Change to Folder sbin: "cd /usr/sbin" and backup old hostapd with "sudo mv hostapd hostapd.bak"
05. Download new hostapd file: "sudo wget http://itwelt.org/downloads/hostapd"
06. Change owner of new hostapd: "sudo chown root:root hostapd"
07. Change permissions of new hostapd: "sudo chmod 755 hostapd"
08. Edit hostapd config file with: "sudo nano /etc/hostapd/hostapd.conf"
09. Replace line "driver=nl80211" with "driver=rtl871xdrv"
10. SAVE & REBOOT
11. After Reboot go to PiPass Webinterface/Dashboard and klick: PiPass > Settings 
12. CHECK if field "Hostapd Driver" contains "rtl871xdrv" and NOT "nl80211". If not enter and save it. A refresh or reboot is required. Also dont't forget to enter your Nintendo's MAC
13. Finally Start PiPass from Webinterface: PiPass > Start and check PiPass logs for any errors.
 
Enjoy

Matthew-Hsu, 2015-05-26:
@schneekluth Thanks for the detailed instructions! I am sure this will help a lot of people with this chipset. Do you know if the new downloaded hostapd works with the nl80211 driver as well? If so, I will follow these steps and release a new pre-made image with this modified hostapd.

schneekluth, 2015-05-26:
@Matthew-Hsu In my case the default nl80211 Driver did not work for me. The logs showed an error stating something like "Unable to start hostapd". I do not remember the exact worrds. I had to switch to rtl871xdrv to get it running. 
Maybe you can simply include the new hostapd in your Image and do a check on boot and use it whenever a 8188cus dongle is detected.

Matthew-Hsu, 2015-05-26:
@schneekluth When I have some time, I will try to follow your steps and see if the hostapd works with the nl80211 driver. If it doesn't, I should be able to rename this modified hostapd with a different name and just use that version when someone enters rtl871xdrv as the driver name.

listix, 2015-05-26:
@schneekluth I tried using your steps but it seems my particular wireless adapter doesn't work even with that. This is the adapter that I have: http://www.vilros.com/raspberry-pi/wifi-adapter.html. My 3DS is simply unable to connect to it. :(

Matthew-Hsu, 2015-05-27:
@listix I wish I could help you out some more with this, but I'll need to see if I can grab that particular chipset to try things out. I downloaded that version of hostapd, but it seems to only work with that chipset and not anything else.

The guide looks about right, but have you tried installing the realtek drivers from: apt-get install firmware-realtek? 

@schneekluth Out of curiosity, are you running the latest version of Raspbian? And did you compile the Realtek drivers from their website or installed the firmware-realtek drivers from apt-get?

Thanks!

schneekluth, 2015-05-27:
@Matthew-Hsu I'm running latest raspian. I tried to install firmware-realtek but it was already the newest version. I havent compiled it myself. I simply downloaded the precompiled Version mentioned in Step 6. So far PiPass does not complain about the new hostapd but i did not have a single mii in my Lobby. Is it normal? How Long does it usually take? Is the database compatible for european 3ds models? Maybe there is still something wrong with the hostapd although PiPass logs say that everything is running fine. I've read a bit more about the rtl8188 chipset and though it all seems to work people say that there remains a problem with the mac spoofing. Most tutorials do not recommend it for the purpose of homepass relays - at least not on Linux. 
I'll receive my new Ralink dongle today and will test wheather I'll have some Streepasses in my 3ds Lobby.

Matthew-Hsu, 2015-05-27:
@schneekluth Oh, I thought you had everything working and was able to get StreetPasses with the RTL8188 chipset. I've had two people who said they got it working, but it was for the original Raspberry Pi. I assume it was running an older version of Raspbian where it is compatible in compiling the Realtek drivers from the website.

But, I agree, the RTL8188 chipset seems to take a bit of work. Often, if hostapd cannot start, there is usually a driver problem. I cannot detect if the WiFi is actually changing MACs. PiPass goes over a checklist and if it can start up all the services along with hostapd, it assumes that it is running fine. In this case, the RTL8188 'should' work, but I guess it falls very short.

The StreetPasses come in quite fast. Generally speaking, when I have my system on standby, it takes about 10 seconds for StreetPasses to hit. The database should be compatible with European models as it contains European zones as well. You might need to personalize the list a little bit more if you want to target European SteetPasses though.

Just a quick reminder, if you'll be switching WiFi dongles on the same installation of PiPass, don't forget to Reset Network under Raspberry Pi. From all accounts, Ralink should work out of the box for you. Just make sure your 3DS is on standby to get StreetPasses.

schneekluth, 2015-05-28:
@Matthew-Hsu Thanks for your reminder. I've received my new Ralink 5370 dongle yesterday and now everything works fine. The first streetpasses came in as soon as I fired up PiPass. If I find the time this weekend I'll create a flashable Image for the __Raspberry Pi Rev.B__ for you.

Matthew-Hsu, 2015-05-28:
Glad everything is working great with the new Ralink 5370!

Creating a Raspberry Pi Rev.B image would be great too! Thanks! 

I'll probably push a minor update this week, so don't feel rushed to create an image. There is another case where hostapd can fail because of invalid MAC addresses, so I just pushed some changes that log extra information, so that others can better diagnose any issues. If you will be using the default PiPass DB list, you won't really see any difference, but if you ever decide to customize it and accidentally input a invalid MAC address into that list, PiPass should be more helpful in those cases.

Again, thanks for all your information regarding the RTL8188 and I hope you enjoy using PiPass!

closed 2015-05-31 by Matthew-Hsu

DeeAce, 2015-06-24:
Not sure if this is the right place for this, please let me know if I should start a new issue, I will give it a go tho.

I have a Wi-Fi dongle (Xtreamer Wi-Fi Dongle) that utilize the RTL8191SU chipset and I'm trying to set it up and get it running with the PiPass installation.

Using the lsusb command I can read that the Wi-Fi adapter is listed as "Realtek Semiconductor Corp. RTL8191SU 802.11n WLAN Adapter".

I have already tried changing the default driver type in the hostapd.conf file and in the GUI to the rtl871xdrv type, but the logs shows that hostapd is unable to start.

It's been a while since I last used Linux, so I'm not really sure where to begin or how to compile and install the drivers from Realtek. Also, what is the difference between RTL8191SU and RTL8191CU?
The SU zip-file does not contain a hostapd file, but the CU zip does (if I need to replace the current hostapd?).

Would be very happy if someone could take a look at this, there are probably more people out there who would like to get Wi-Fi adapters with this chipset to work.

Matthew-Hsu, 2015-06-26:
@DeeAce 

Sorry for the late response, it's been quite busy for me lately. I did some quick research regarding your WiFi adapter and it does seem that people are having problems with it running hostapd correctly.

Realtek WiFi adapters seem to have lacking Linux driver support in general. Even with hostapd support, you may run into the issue of having a WiFi adapter that is unable to change MAC addresses, which is an essential component to Homepass.

You could still try it out. The hostapd for the CU variant should work with the SU variant. I do not own this WiFi adapter, so it is quite difficult to give clear instructions, but in general, you will want to replace your current hostapd with the modified Realtek hostapd. Once that is done, you'll have to use PiPass Settings and set the correct hostapd driver name. If you use PiPass Update to grab the latest changes, PiPass should be able to tell whether your WiFi adapter is unable to change MAC addresses at that point.

DrakePt99, 2017-11-04:
Hi, First of all i need to introduce myself. Im António, im from Portugal and i need to tell you all one thing. I have a Chronic Psiquic Disease which dont let me go outside work, meet people and make friends, I learn to live with that these years. I pass my days in my home playing games and watching TV. I love games, i got my first console when i was a child :). This is my story and sorry to tell you this.

So, about streetpass, i purchased the plaza games but i cant play them, because i dont get streetpasses, thats why i ask for help here.

I want to build my own Homepass with a Raspberry Pi, there anyone experienced to help me build one?
I don´t have any friends and i dont know how to make one Homepass. Help me please.

I have a Discord account, twiter and facebook, we can talk by these or here.
I promising not be annoying okay?

I hope for your help guys.
i wait reply.

thanks to all