Matthew-Hsu/PiPass #23: Removing the Mac Address Whitelist

fewyn, 2015-06-02:
Is there a way to remove the whitelist? I'm planning to set up a PiPass and would like to be able to let any 3ds to connect to it.

nagledb, 2015-06-02:
I'm not sure if it's plausible or not, but be aware that you wouldn't just be letting any 3ds connect. That would allow any device to connect, including passers-by with mobile devices and neighbors with computers in range. You'd be sharing your internet connection openly with the world.

fewyn, 2015-06-02:
That's fine for the location the device is actually going to be in (a local game store)

nagledb, 2015-06-02:
Well, that sounds pretty awesome. :+1:

Matthew-Hsu, 2015-06-02:
@fewyn I never thought about a case like this and I think it sounds pretty awesome. I can definitely add enabling/disabling security in PiPass Settings. It's an easy change and I can probably add this later tonight.

If you're comfortable in Linux, the actual change is quite easy. If you open up piPass.py in /opt/PiPass/, go to line 340 and change the value of macaddr_acl=1 to macaddr_acl=0. The entire line should look like this after you made your changes:

conf = "interface=wlan0\nbridge=br0\ndriver=" + HOSTAPD_DRIVER + "\nssid=" + zoneValues[0] + "\nbssid=" + zoneValues[1] + "\nhw_mode=g\nchannel=6\nauth_algs=1\nwpa=0\nmacaddr_acl=0\naccept_mac_file=/etc/hostapd/mac_accept\nwmm_enabled=0\nignore_broadcast_ssid=0"

Hope this helps. I will let you know when I finish the changes when I get around to it if you want to wait.

Matthew-Hsu, 2015-06-03:
@fewyn Disabling security has been implemented if you want to check it out. I've updated the pre-made images with these changes if it makes things easier. If you are using PiPass Update, check out the readme for some more information regarding this change.

Hope this works out nicely for your store! Thanks!

closed 2015-06-03 by Matthew-Hsu