Matthew-Hsu/PiPass #24: Providing combined JSON stats

schneekluth, 2015-06-03:
Hello @Matthew-Hsu ,
I'm quite fascinated by PiPass. Currently I'm waiting for my Pebble Time and I played a bit with Pebble's Cloudpebble IDE. I've just started with programming and thought it might be a nice beginner's project to write a Pebble App for PiPass using pebblejs. Basically I'd like to display stats on the tiny display but I've also other ideas like "shake your pebble and advance to next zone" or change settings like cycle times.

I've seen, that you provide three JSON-Files in _'var/www/assets/json'_. I managed to perform AJAX calls on the _pipass_config.json_ and extract it's data but I have a problem doing the same with _current_state.json_. I think it has to do with the json keys because they contain a dollar sign e.g. *gsx$ssid:* or *$t:* (from the google spreadsheet). I tried using escape character but couldn't manage to extract the data. There is also data missing that I'd like to display like the current version of PiPass that is running.

Therefore I'd like to ask wheather you can provide a JSON file in an upcoming version that combines the pipass_config data and the current_state data but without the Dollar sign in the keys name just like in _pipass_config.json_ and with the current version number ? This way I could retrieve the data with just one AJAX call in pebblejs.

This is what it looks like right now:
[image]

Matthew-Hsu, 2015-06-05:
Hey @schneekluth ,

Your Pebble app looks really great! The shake feature is pretty neat and I would imagine it would work well for StreetPass sessions.

As for the JSON data, I'm curious, if you were to manually edit current_state.json with JSON keys that do not have the '$' in them, are you able to query the data? I haven't used pebble.js, but do you know if they support querying of nested JSON? Google provides an API to use and they send back JSON with nesting involved. It is possible to clean that data up, but it may require an additional step in retrieving the Google provided nested JSON by parsing the required data and reassigning it a new, non-nested, key.

I'm not sure about combining pipass_config.json and current_state.json. The additional writes are OK since PiPass will write to current_state.json anyways, but I would have to load the settings each time and encode them back to JSON through each zone traversal. I can, however, include the versioning number in one of those files for you to parse.

We can discuss some solutions here too because your Pebble app looks awesome! But, check out the JSON nesting though, there might be a way to access the nested values that you need.

Thanks!

schneekluth, 2015-06-05:
Hello @Matthew-Hsu ,
Good news. I managed to get the AJAX call to work with the original current_state.json including the Dollar signs. Therefore everything is fine now and the data get's properly extracted and displayed (except the missing version number). Keep up the good work!

closed 2015-06-05 by schneekluth

Matthew-Hsu, 2015-06-06:
@schneekluth , that's good to hear! I also wanted to talk to you about the versioning. I do not release a version change each time when PiPass is updated. If, for example, I push something on GitHub, I would usually state that this version is running on master branch updates that are based on 1.X. For your Pebble app, I could do something like 1.52M to indicate that the version is based on 1.52, but using some master branch additions to it.

If you like something like this, I can go ahead and add the versioning. Since most of the JSON files are being changed, if I included another JSON file from where you can pull from, would that be OK? I would add it to /var/www/assets/json/ and I can make sure that PiPass does not touch that file and maintain the versioning number there when I do releases.

Let me know what you think and I will definitely try to work with you to get your watch app out!

schneekluth, 2015-06-06:
@Matthew-Hsu I think the idea with an extra JSON with an indicator like 1.52M is great. I could perfectly use it that way. And also thanks for supporting my idea of a Pebble app. Learning JavaScript that way is great for me as a beginner.

Here is an update of the watchface. I used a smaller logo so there's enough space for the data. 
[image]

Matthew-Hsu, 2015-06-06:
It looks really great! If I had a Pebble, I would have liked to try it out.

I just pushed versioning to /var/www/assets/json/pipass_version.json for you to use. Let me know if this works out for you or if you need anything else.

Depending on how you feel, if you want a shout out on the readme, I don't mind linking to your repository.

schneekluth, 2015-06-07:
Hello @Matthew-Hsu , pipass_version.json works perfectly. At present I don't think it's the right time to promote the app in your README because there are too many thinks missing. I'll add a menu for controlling PiPass next weekend and see if I can find someone who lends me his Pebble to test the app. I'd suggest to finish the basic structure and functions until we mention it in your repo.

Matthew-Hsu, 2015-06-07:
That is great that everything is working fine now.

Sounds good with the app, I look forward to seeing it!

LoganMcClay, 2015-06-09:
Hi ! Hey @schneekluth I'm also a PT Kickstarter owner, still waiting for it too, can't wait to receive it and see how it performs with your app and PiPass ;)