Matthew-Hsu/PiPass #25: driver issue/hostapad

xthedakmanx, 2015-06-08:
2015-06-08 10:53:13,951 - __main__ - INFO - PiPass is now running.
2015-06-08 10:53:32,508 - __main__ - WARNING - A possible WiFi driver issue has been detected.
2015-06-08 10:53:32,510 - __main__ - ERROR - Unable to start hostapd.
2015-06-08 10:53:32,512 - __main__ - INFO - PiPass has been shutdown with an error.

=\

using http://www.amazon.com/gp/product/B008XVAVAW/ref=as_li_tl?ie=UTF8&camp=1789&creative=390957&creativeASIN=B008XVAVAW&linkCode=as2&tag=matthew08b-20&linkId=IN557BBG3PJEAU7M

nagledb, 2015-06-08:
A few ideas to try if you haven't already:

Try rebooting the system. If that works, then maybe it didn't pick up the wifi dongle when it booted for some reason.

Try moving the wifi dongle to a different usb port. (Be sure to reboot after, it won't automatically detect changes I don't think.) If that works, then perhaps one of your usb ports has an issue.

xthedakmanx, 2015-06-08:
Fixed it. Incorrectly put in the MAC address. Thanks ^_^

closed 2015-06-08 by xthedakmanx