Matthew-Hsu/PiPass #26: Pipass stops cycling on network interruption?

Corruptinglyneedful, 2015-06-10:
I seem to have noitced that Pipass will stop cycling if there is a network interruption or connection drop.

A few days ago I had a network hiccup and the connection d/c'd for about 2 minutes.  Pipass wasn't working after the connection had been re-established.

I had to manually restart everything.

I can understand doing this for a power outage, but I can't understand it for network interruption.

Is this working as intended?

Matthew-Hsu, 2015-06-10:
I would probably say the issue is in a grey area. If you can, could you post snippets of the log just before PiPass terminates? I did a quick disconnection test by unplugging the ethernet from my Pi and PiPass will continue cycling without an internet connection. I am guessing it is terminating when it is trying to download the Nintendo Zones from the database upon start or when it reaches the end of the current list or when a refresh is initiated? Here is an example:

2015-06-10 03:25:58,291 - __main__ - INFO - PiPass is now running.
2015-06-10 03:26:18,334 - __main__ - ERROR - Unable to read the URL: https://spreadsheets.google.com/feeds/list/1OfgyryUHeCPth76ziFT985XNLS-O5EXtjQDa0kA1L6M/7/public/values?alt=json.
2015-06-10 03:26:18,337 - __main__ - INFO - PiPass has been shutdown with an error.

With the database stored online, the behaviour does make sense in this case. Depending on the logs, we can see how we want to handle this. We could pause and retry, but we probably don't want to retry until a valid connection is reached.

Matthew-Hsu, 2015-06-12:
@Corruptinglyneedful , I have just issued an update that will check network connectivity and attempt to reconnect up to five times before exiting out. This entire process will last for about 5 minutes, so any extended network outages will require you to start PiPass up again.

PiPass just needs Internet access when pulling information from PiPass DB. It makes this request when PiPass initially starts up, refreshes, or when it reaches the end of the list. Once the information is loaded, it will continue to cycle through MACs, but since there is no Internet access, Homepass will not work anyways.

The reconnection will help alleviate any issues you were having, but it won't try forever until a Internet connection is established.

Corruptinglyneedful, 2015-06-12:
OH, sweet.  How nice of you to actually do that, as that's more a quality of service kind of update.

That's pretty nifty, yes obviously it couldn't try forever but 5 minutes should be plenty of time to reconnect even if it's the last device on a network.

Thanks Matt, I'll update right away to test it.

Matthew-Hsu, 2015-06-12:
No worries! It was easy to implement and if it makes your experience better, then it's good! If five minutes does not make sense in the future, this is something that we can revisit and change quite easily.

Thanks for the feedback, it's always appreciated.

Corruptinglyneedful, 2015-06-12:
I'm wondering why there has to be a time limit at all?  Why it couldn't just constantly check to make sure there is a network connection every so often, indefinitely?  

I'm assuming it's to keep it minimal, so it's fast, snappy, and responsive.

Hmm, wish I could get others feedback on it.

Matthew-Hsu, 2015-06-12:
A terminating state is always good to have. I did do a quick test and stopping PiPass through the GUI can still stop PiPass if it tries to reconnect indefinitely.

As I see it now, the change itself should be easy without introducing so much overhead that it will slow PiPass down.

Since the current update takes care of case 1 where PiPass needs to download the database information, the other case would be where it already has the information, but it loses internet connection. Since Homepassing will not work, I could pause the cycle and resume where we left off when a internet connection has been established again.

The change is quite easy to do, but it would require a bit of long term testing since it could break PiPass.

Matthew-Hsu, 2015-06-12:
I've been running PiPass for about 2 hours and it seems OK with the changes we talked about. It will try reconnecting indefinitely and will check network connectivity throughout.

When a network interruption has been detected, PiPass will halt cycling and resume when a connection has been established again. So far, all the other functionalities seem to be working with the changes. Right now, I'm just running it overnight to ensure longevity is unaffected.

There is no one click button to upgrade to this yet, since I want to test first, but if you want to check it out, you can download piPass.py from here: https://github.com/Matthew-Hsu/PiPass/tree/BETA/opt/PiPass

and overwrite it with piPass.py in /opt/PiPass/ on your RPi. If you do this, you can always revert back to master by clicking PiPass Update.

Corruptinglyneedful, 2015-06-12:
What would I be testing exactly?

Matthew-Hsu, 2015-06-13:
Surprisingly enough, my ISP had some issues today and PiPass handled it quite well. It did not stop and appropriately stopped cycling and resumed after network connectivity.

Determining a valid network connection is a little bit tricky. I am using two public DNS servers in which I ping to determine if we have a Internet connection. These are the Google public DNS servers, so they should be reliable.

I have updated piPass.py with these new changes and I suppose additional help with testing would be great if you don't mind. Just use PiPass as you normally would and report any oddities that you may find. I am confident that the dropped connection / reconnection parts work. I just want to ensure that existing functionality is not broken with the changes.

Thanks!

Corruptinglyneedful, 2015-06-13:
Well now, isn't that something.  No better way to test my theory than to have an ISP issues.  That's what we call good timing on my part.

So I can Update through the Dashboard now?

Matthew-Hsu, 2015-06-13:
I just left PiPass running overnight to make sure it is stable. The changes should be good to try out now. Using PiPass Update will give you these changes now.

closed 2015-06-13 by Matthew-Hsu