Matthew-Hsu/PiPass #27: Can't configure PiPass!

FamicomGuy, 2015-06-10:
I tried installing the image onto my SD card for my Model B Raspberry Pi, but I can't do much without a GUI, which I can't install because I can't get it to connect to Wifi no matter what I do!

And installing it manually is a lot of work and it denies me permission to add the PiPass_master files when I connect to my Pi via WinSCP.

Can you find me a solution to either of these problems?

Matthew-Hsu, 2015-06-10:
Hey,

Can you just verify some things for me? I remember in another thread you were saying something about having a RPi 2 Model B, do you actually have a RPi 1 Model B?

If you're talking about the RPi 2 Model B, check the following:

* RPi is connected to your router via ethernet.
* RPi has the wireless USB dongle plugged in.

If these things look good, get the IP for your RPi. Once you have that, enter the IP of your RPi into a web browser on another device. You should be able to see the PiPass GUI.

If you're talking about the RPi 1 Model B, there is no pre-made image for that and you must do a manual install. If you're running into permission issues, you should do all your remote work under the user: root. You won't run into any permission issues. If you're using the default Raspbian OS from Raspberry, then you need to configure root access first. Run this command on your Raspberry Pi:

* sudo password root

Give root a password and use the user / password combination for all remote work through WinSCP.

FamicomGuy, 2015-06-10:
Yes, I have the RPi 2 Model B, and yes that was me at GameFAQs.

I do have the dongle since I bought the starter kit, but when I connect my ethernet cable into my RPi from my computer, my computer thinks its an unidentified network with no Internet even though my router shares its connection, therefore the indicator on the dongle doesn't light up.

FamicomGuy, 2015-06-10:
Never mind, I got it to work! I connected my RPi directly to my router via ethernet and it works!

FamicomGuy, 2015-06-16:
I've gotten the most out of PiPass this last week. But now I want to install it manually to an SD Card that  initally has NOOBS on it so I can have it on a Pi instance that has a proper Raspian GUI once I set a root user on it.

I thought about simply copying the var, opt, and etc files on the image with PiPass already installed and merging them with the ones on the NOOBS image. Would this work?

closed 2015-06-16 by FamicomGuy