Matthew-Hsu/PiPass #29: Using two USB dongles?

weiweicheng, 2015-06-11:
Is it possible to setup a second USB dongle instead of ethernet to gain internet access? How would I go about doing this?

Matthew-Hsu, 2015-06-11:
I suppose it would be possible, but I have never tried this before. I don't have a second WiFi USB dongle, so I won't be able to try it out, but I can give some tips:

* The first USB dongle that you will plug in should be your wireless access point that will broadcast the Nintendo Zone.
* The second USB dongle will replace the Ethernet connection and serve as providing Internet access to your RPi. So, you will want to verify that wlan1 is providing internet access to your RPi.
* You'll need to modify /etc/network/interfaces and replace all instances of eth0 to wlan1.

Again, I'm not too sure if these instructions will work without some tinkering.

weiweicheng, 2015-06-11:
Would I also have to add these two lines to /etc/network/interfaces in addition to changing the bridge from eth0 to wlan1?  I am basing this on the master branch version of the file. 
>allow-hotplug wlan1
>iface wlan1 inet dhcp

Can I also get this to work with wpa_supplicants.conf to retain my SSID settings?

Matthew-Hsu, 2015-06-11:
Yes, you'll have to add those as well. I found a thread here that may be relevant: http://raspberrypi.stackexchange.com/questions/12276/how-to-make-pi-an-ap-with-2-wifi-adapters

PiPass will use wlan0 as the actual access point, so you do want to ensure that wlan1 replaces the function and duties of eth0. PiPass only writes to hostapd.conf and will not modify wpa_supplicants.conf, so your settings should be retained.

closed 2015-06-14 by Matthew-Hsu