Matthew-Hsu/PiPass #30: PiPass won't cycle or log won't write

FamicomGuy, 2015-06-11:
Today I woke up and PiPass's log says it's running but it won't switch to a MAC address, or at least the log file doesn't say so because I got a streetpass despite this. I tried advancing, refreshing, and restarting PiPass, but it won't go. And I have my Google Doc set to cycling the Nintendo Zones [2].

Matthew-Hsu, 2015-06-11:
I'm curious, could you use another device or clear your browser cache and try accessing the dashboard again? If you get a page not found, it could be a case where your router has assigned a new IP to your RPi and you are using a cached version of the webpage. If you're still getting passes, PiPass is probably still running in the background.

Another thing to try is to manually log into your RPi and type: cat /opt/PiPass/logs/piPass.log

Check the timestamps and see if it is still running.

Another thing, could you take a screenshot of your PiPass settings?

FamicomGuy, 2015-06-11:
It is running, but it won't switch to a MAC Address or log any refreshes or advances. The only thing it logs is that it is running. And the IP address is the same. And it could be true that it's still running in the background. The only timestamp I get is that it is running. And here is a screenshot of my PiPass settings:

[image]

The DB key is that of my own personal Google Doc. I copied the contents of the second worksheet (Nintendo Zone MACs) over to the first.

FamicomGuy, 2015-06-11:
It is running, but it won't switch to a MAC Address or log any refreshes or advances. The only thing it logs is that it is running. And the IP address is the same. And it could be true that it's still running in the background. The only timestamp I get is that it is running. And here is a screenshot of my PiPass settings:

[image]

Matthew-Hsu, 2015-06-11:
I saw that you removed your last screenshot. Are you using your own PiPass DB? Did you make sure that you published your own PiPass DB as public after creating it? Check back with the documentation to make sure you did this step.

In order to parse data from a spreadsheet, you do need to make this public. This does not mean that people can edit it, so your list will be fine.

If you want, send me a direct link to your PiPass DB. I should be able to view the content, but not edit it.

FamicomGuy, 2015-06-11:
Yes, I'm using my own DB, and I did publish it to the web. Here's a link:

https://docs.google.com/spreadsheets/d/1yJowMAPGbpaIA_58mPQyaOhicUhIrXmKZE_5ybkAeb4/pubhtml

Matthew-Hsu, 2015-06-11:
I just checked it out, this may look quite gibberish, but I did notice a problem:

This is the original PiPass DB: https://spreadsheets.google.com/feeds/list/1OfgyryUHeCPth76ziFT985XNLS-O5EXtjQDa0kA1L6M/1/public/values?alt=json

And yours:
https://spreadsheets.google.com/feeds/list/1yJowMAPGbpaIA_58mPQyaOhicUhIrXmKZE_5ybkAeb4/1/public/values?alt=json

Your PiPass DB is assigning different keys from mine and I am guessing PiPass cannot parse that correctly. For example, gsx$ssid should always use this key. Your worksheet #1 is doing this: gsx$ssid_1, gsx$ssid_2, etc.

Could you revert back to the original PiPass DB and see if this problem goes away? I don't think cell colouring should make Google rename the keys, but it is something that you can try. So, try to keep the formatting of PiPass DB as original as possible. Of course, the zone content can be different.

You might want to restart your Pi as well.

FamicomGuy, 2015-06-11:
It worked! Thanks! Was the problem because I color coded the DB?

closed 2015-06-11 by FamicomGuy

Matthew-Hsu, 2015-06-11:
I'm guessing so. I guess Google sees the difference in colour and encodes it differently. It is a strange case, but at least we know what is going on.

I'll play around this and maybe I'll be able to catch this case and output something meaningful in the future.

Glad it is sorted out now!