Matthew-Hsu/PiPass #31: Streetpassing

FamicomGuy, 2015-06-11:
I noticed PiPass only works in sleep mode. Is this because of the hostapd security that only sends streetpasses to authorized 3DS systems? While I am aware that turning it off allows all 3DS Systems to receive passes, does it also allow them to receive passes while active?

Matthew-Hsu, 2015-06-12:
It's not PiPass specific, but Homepass solutions will work only in sleep mode.

closed 2015-06-12 by FamicomGuy

Corruptinglyneedful, 2015-06-16:
@FamicomGuy  You can get Street Passes while your 3ds is open over Homepass, it just needs to not be connected to a nearby network.

The 3DS is incapable of having a network connection and getting street passes at the same time.

I d/c'd my 3ds from my network and I can get passes while playing.

FamicomGuy, 2015-06-16:
That worked! Thanks!

Corruptinglyneedful, 2015-06-16:
Glad, I could be helpful.  Now I can say I actually gave back to the community.

nagledb, 2015-06-16:
Aha! I almost always play my 3DS at home, so I had that issue too. Thanks @Corruptinglyneedful !

Corruptinglyneedful, 2015-06-16:
@matthew-hsu, look at me being useful and shit :D

Matthew-Hsu, 2015-06-17:
@Corruptinglyneedful Awesome, thanks for the tip! Never really noticed this until now. I'll update the readme with this information.

Corruptinglyneedful, 2015-06-17:
Sounds good.  I think this catches everyone at some point.  It's not something you'd think about normally.