Matthew-Hsu/PiPass #32: Invalid MAC Addresses

FamicomGuy, 2015-06-13:
I was trying some new MAC Addresses with PiPass in an effort to set up my own private MAC, but it seems some MACs are invalid and can't be used with PiPas due to hostapd not recognizing them. Do only the MACs already in the seven worksheets of the PiPass DB work for PiPass? Is there a way to alter hostapd to accept more types of MACs? Or is it Homepass itself that won't work with other MACs?

nagledb, 2015-06-13:
One thing I did notice is that hostapd seems to expect mac addresses to be colon separated instead of hyphen separated. (For example, some of the addresses in the [Personal Mac's sheet](https://docs.google.com/spreadsheet/ccc?key=0AvvH5W4E2lIwdEFCUkxrM085ZGp0UkZlenp6SkJablE#gid=6) of the main Homepass spreadsheet are hyphen-separated; an example: 00-06-4F-84-65-12.) I manually fixed the ones I copied to my personal spreadsheet, but that's something that could potentially be addressed in PyPass by replacing hyphens with colons.

There's also some apparently-valid mac addresses that hostapd just won't accept, like CB:8E:19:6B:D1:F2. I have no idea why.

Matthew-Hsu, 2015-06-13:
Yes, PiPass is limited with what hostapd can accept as a valid MAC address. Like the case that nagledb talked about, I'm not sure why hostapd won't accept a seemingly valid MAC address.

To make things easier, I can add a step that will replace all '-' with a ':' instead.

FamicomGuy, 2015-06-13:
I don't know if that's necessary since Google Docs has a "Find and Replace" option, but if it's possible to alter the DB connected to PiPass outside of Google Docs in that way, then feel free to do it.

Matthew-Hsu, 2015-06-13:
It won't alter the DB in any way. It will just change it internally when all the data is saved into PiPass. The nice thing about this change is that it's about 12 characters of code to do the replacement, so it's very easy to do and check for.

FamicomGuy, 2015-06-13:
Ok, if this means PiPass will accept both colons and dashes between hex digits, then that's an improvement.

Matthew-Hsu, 2015-06-14:
If you update PiPass, you'll get these changes now.

closed 2015-06-14 by Matthew-Hsu