Matthew-Hsu/PiPass #33: hostapd security issue

perryhuynh, 2015-06-16:
After enabling 'hostapd security', any client can still connect to the network

nagledb, 2015-06-16:
It's normal for it to broadcast its SSID, even with hostapd security enabled. But it should prevent clients that aren't on your "3DS Authentication" list from connecting.

I just updated to the latest version and tested it out. With security enabled, I'm not able to have my PC connect to the PiPass attwifi network. Did you try having a client that isn't in the "3DS Authentication" list connect to see if it actually lets it successfully connect?

perryhuynh, 2015-06-16:
Yes, I can still connect to it using any computer/device in my house

nagledb, 2015-06-16:
Hmm. Maybe @Matthew-Hsu will have some ideas, but in the mean time, a few things I can think of you might check:

Do a force refresh on the settings page. (Maybe it's showing you old settings?)

Try using the "Refresh" option to refresh its settings. (Maybe it didn't actually pick them up yet?)

Try restarting PiPass. (Maybe it's still running an old version of the client and is confused?)

perryhuynh, 2015-06-16:
I've tried both, going to SSH into the RPi and check the config files

closed 2015-06-16 by perryhuynh

Matthew-Hsu, 2015-06-17:
I double checked the functionality and it seems to be working correctly for me. Hostapd security will take effect in the next zone and not in the current one, despite refreshing, since writing new values to hostapd.conf only happens at the start of a new zone.

PiPass will stop broadcasting the access point when stopped. Depending on your device, I do know some devices will cache the wireless networks. MacOSX seems to do this, but when using my Android phone, I can see the network disappear shortly after PiPass has stopped.