Matthew-Hsu/PiPass #34: PiPass + Pi Camera Question

FamicomGuy, 2015-06-16:
The other day I bought a 5MP Camera that can be attached to the Raspberry Pi. Can it be used with a Raspberry Pi that has PiPass on it? Since PiPass SD Cards also have Raspian and raspi-config installed, I assume the camera can still be used with it. Or do I have to use an SD Card that has the normal NOOBS image?

nagledb, 2015-06-16:
I haven't tested it out (don't have the camera), but I can't imagine why it wouldn't work. Try it out and see. :)

The instructions for it are here:
https://www.raspberrypi.org/documentation/usage/camera/README.md

If you don't see the camera option in the raspi-config menu, you may need to update your packages. This page mentions how to do that:
http://thepihut.com/blogs/raspberry-pi-tutorials/16021420-how-to-install-use-the-raspberry-pi-camera

FamicomGuy, 2015-06-16:
I think it should work, because I logged into my Pi's command line remotely, and it does have raspi-config with the camera option on it. I'm gonna go try it out.

FamicomGuy, 2015-06-16:
The camera works!

closed 2015-06-16 by FamicomGuy

Matthew-Hsu, 2015-06-17:
Awesome, glad it is working! I updated the readme to add another installation method. I have created an installer that will automatically install PiPass on a Raspbian based operating system. Someone on GBATemp has successfully used it on his old RPi and it should make installing PiPass easy for those who do not want to use the manual method. The installer will use default paths, so if you need extra customization, the manual install guide is still the best to follow.