Matthew-Hsu/PiPass #35: IP Address Port Number

FamicomGuy, 2015-06-17:
How do you change the port number (80 by default) that PiPass runs on in case I want to install a new utility for my Raspberry Pi that is accessed by entering the Pi's IP into the address bar? Can I do the same for any other utility? (This is to prevent overriding PiPass's pages)

nagledb, 2015-06-17:
PiPass just uses the default Apache installation. It's fairly straightforward to have it work on a different port. There are two files you'd have to modify. The first is /etc/apache2/ports.conf. You should see these two lines:

    NameVirtualHost *:80
    Listen 80

Change 80 to whatever port you want to use. The second file is /etc/apache2/sites-enabled. The first like looks like this:

    <VirtualHost *:80>

Change that 80 to match your port as well. Once you've edited both files, run this command to reload the configuration files in apache:

    service apache2 reload

You'll probably get the message "Could not reliably determine the server's fully qualified domain name, using 127.0.1.1 for ServerName". You can ignore that, it's normal. As long as it says "ok" and doesn't give any other error messages, you should be able to connect on the new port.

As long as your other utilities don't use Apache, you should be set. However, if they do use Apache, then you may need to learn a bit more about how to configure Apache so that the services can all coexist within Apache on separate ports.

FamicomGuy, 2015-06-17:
I installed another utility that uses Apache, and now I can't access PiPass! It must of overwritten it or something. Does anyone know what to do about this, regardless of what the other utility may be?

closed 2015-06-17 by FamicomGuy