Matthew-Hsu/PiPass #36: Log looks good, but no streetpasses

SeKirch, 2015-06-17:
Hi, i have a Raspberry Pi 2 and installed Pipass. Anything looks good. The log say PiPass is now running and then Spoofing as... messages. I have tested with Hostapd Security on and off. i can connect from my smartphone and can browse in the internet. But i get no streetpasses. What do i wrong?
Any idea?

Thanks
SeKirch

nagledb, 2015-06-17:
If you can connect from your smartphone and it's showing that it's spoofing in the log, then it sounds like everything on PiPass's side is working. I hope this doesn't sound like a dumb question, but are you sure your 3DS is set up to receive StreetPasses? Have you received any non-PiPass StreetPasses on it?

Alternately, are you sure your 3DS games have room for additional StreetPasses? Each game has a limit on how many StreetPasses it can hold. Once full, you need to play the game and use them so it clears up room. For example, Mii Plaza only has room for 10.

Nintendo has an article on StreetPass that may be helpful:
http://en-americas-support.nintendo.com/app/answers/detail/a_id/681/~/tips-for-setting-up-streetpass

SeKirch, 2015-06-17:
Yeah streetpass is activated. i have two 3DS and they meet each other. Additionally i tried the last two days a method with a WIndows 8.1 PC and another WIFI Dongle (TP Link) and get streetpasses over this method. But the PiPass method is more comfortable.
One thing looks curious. In my fritzbox i see the Pipass connected over LAN, but with a comment that it is connected over my pc.
My setup is that from my fritzbox goes one ethernet cable in a switch and the pi and pc etc are on the switch. Can this be a problem?
Thanks so far

nagledb, 2015-06-17:
I'm not familiar with Fritzbox. Looking at it on Wikipedia, I don't see any reason why that would be the source of issue, but I can't say for sure.

When you connected with your smartphone, was it able to access the Internet? If PiPass can't reach the internet, then it wouldn't be able to get data for StreetPass. If it's somehow connecting to your PC for a network, then maybe it's on a local-only network and can't see the Internet.

SeKirch, 2015-06-17:
Yes my smartphone has internet acces when i connect to e.g. attwifi from pipass. But my 3DS cannot connect to the attwifi network. When i search with my 3DS for Acces Points and choose them, the connection test fails at the connection to the Acces point

GCSkater, 2015-06-18:
I am currently having a similar problem. I have 2 3DS's both setup for streetpass properly. PiPass is setup correctly, and logs are showing that it is Spoofing correctly. However, i have not gotten any streetpasses.  I have Advanced the PiPass to a new MAC address and the log shows that it changes correctly, but it still doesn't send streetpasses. I am able to connect my phone to the wireless signal coming from the Pi, and I am able to browse the internet through it (which it shouldn't because i didn't add my phones mac address to the list right?)

I ran a 'ifconfig' to see if the adapter was actually changing the MAC address, and wlan0 shows its connection as Ethernet with the MAC address as the actual Pi MAC address (not the spoofed one).

Shouldn't wlan0 show up as the Spoofed MAC address? Anyone know what changes i need to make to make sure the MAC address gets spoofed correctly?

Any help is appreciated.

SeKirch, 2015-06-18:
Can you connect with your 3DS to the wireless signal from the pi and browse the internet?

GCSkater, 2015-06-18:
I set up the connection and was able to go to the eshop and google a couple of things.

Matthew-Hsu, 2015-06-18:
Could you list the model of the WiFi dongle that you are using and if possible, the wireless chipset as well?

wlan0 ought to be matching the MAC of the spoofed zone. The only time I have seen this is in cases where the WiFi chipset does not support the changing of MACs. In these cases, the access point can still function as providing Internet access and hostapd will be happy and still run.

GCSkater, 2015-06-18:
Hey Matt, 

I am using this Wifi Dongle: EDIMAX EW-7811Un, Realtek RTL8188CUS. I was able to change the MAC address before when you could still set up Homepass through windows, so I know the adapter works. 

I was originally having Wifi drivers issues, but then I found this post by schneekluth:

https://github.com/Matthew-Hsu/PiPass/issues/20

I followed the instructions so that cleaned up the driver error, but still no passes. 

Now i did notice that the 'Spoofing Status' on the dashboard does say that the MAC Address and the SSID are not available and that the PiPass is not running, but the logs says that the Spoofing has started. I am unsure if that helps or not, but i figure any information i can provide would be beneficial. Let me know if you need anything else.

Matthew-Hsu, 2015-06-18:
@GCSkater 

RTL8188CUS is quite problematic with hostapd. Even though it works with Windows, the drivers for Linux do not seem to be polished as such. I believe schneekluth got up to your point and had a working access point, but also could not get hostapd to change MACs. I believe he eventually switched out to a Ralink chipset.

The logs will generally be more up-to-date than the dashboard. With a web interface, sometimes browser caching will kick in and will require a manual refresh. Generally, if you see in the logs that it is running and do a manual refresh of the browser page, the dashboard status should reflect the logs. After the manual refresh, browsers will often start pulling the latest data.

Matthew-Hsu, 2015-06-18:
@SeKirch 

If you are using the supplied network configuration with PiPass, it will rely on DHCP. The IP of your RPi should be stable, even when switching zones. Many routers will do this, but I do know there are cases where the RPi will be assigned a new IP at each zone switch. There should be a configuration with your DHCP that can provide a relatively stable IP. If you cannot find this, you could try to provide a static IP to your RPi instead.

This guide is quite helpful and you will want to pay attention to 'Configuring the Network Interface'.

GCSkater, 2015-06-18:
Thanks for the quick reply Matt. 

Logs were definitely more up to date then the dashboard, I cleared browser cache and everything updated.

I was hoping this wireless adapter would come through for me, but it doesn't look like it's a simple settings error. I'll be sure to use your referral link for the new adapter. Thank you for the help Matt!

Matthew-Hsu, 2015-06-18:
@GCSkater 

No worries! Just a quick reminder, if you will be switching between WiFi adapters on the same installation of PiPass, you may want to use the 'Reset Network' option.

In the future, I should be able to add a MAC verification step to print out some messages when the WiFi adapter is unable to spoof correctly.

SeKirch, 2015-06-19:
After cleaning up my router, restart the 3DS and reinstall Pipass on my Raspberry it now works perfectly. Thanks for the help and the very cool program. Good job ;)

Matthew-Hsu, 2015-06-20:
Awesome! Glad it is working out for you. Enjoy it!

closed 2015-06-20 by Matthew-Hsu