Matthew-Hsu/PiPass #37: Problem with hostapd

FamicomGuy, 2015-06-19:
After I decided to start with a fresh instance of Raspian on my Pi (I have an image backup) to test the new batch script, after I ran it on my pi and restarted it, it changed the IP address associated with its Ethernet connection, and it won't run because hostapd keeps failing.

I didn't have this problem at all when I installed it manually or used the pre-installed image, though. Does anyone else have this problem after installing PiPass with the bash script?

Matthew-Hsu, 2015-06-19:
Since the IP is dynamic, in cases where you will be modifying your operating system environment, you may be assigned a new IP from your router.

If Hostapd fails, it is generally something with your installation of the WiFi driver. It seems you are changing your environment quite a bit, if you can, could you use the 'Reset Network' option? Give it a quick read about it on the readme before you try it out.

FamicomGuy, 2015-06-19:
I just figured out the IP address change was with the dongle I used, and hostapd not working has to do with not setting the right permissions.

Matthew-Hsu, 2015-06-19:
What permissions did you need to set with hostapd?

FamicomGuy, 2015-06-19:
I almost have it working again, but the log keeps giving me this:

2015-06-19 04:50:22,530 - __main__ - INFO - PiPass is now running.
2015-06-19 04:50:43,447 - __main__ - WARNING - A possible WiFi driver issue has been detected.
2015-06-19 04:50:43,448 - __main__ - ERROR - Unable to start hostapd.
2015-06-19 04:50:43,451 - __main__ - INFO - PiPass has been shutdown with an error.

FamicomGuy, 2015-06-19:
I think it's because I switched the dongles on my Pi (I bought a better one) that caused hostapd to stop working properly. If that is the case, how do you reconfigure hostapd to accept the new one and the IP it provides?

Matthew-Hsu, 2015-06-19:
What is the model of the WiFi dongle and if possible, what chipset is it based on? When purchasing WiFi dongles, you have to ensure that it can act as an access point and it is able to change MAC addresses as well. Since you are getting unable to start Hostapd, there is a possibility that the WiFi dongle you are using is incompatible with Hostapd.

Best bet is to Google search your WiFi chipset and verify the hostapd driver name. If it is different than the default one supplied in PiPass, you may have to change that name. Also, if you do get to the point where hostapd does start up and you are still not getting StreetPasses, you'll have to verify that your WiFi dongle is actually changing its MAC address to match that of the current Nintendo Zone.

FamicomGuy, 2015-06-19:
I'm now using an Edimax EW-7811Un 150Mbps 11n dongle, with the chipset of Realtek RTL8188CUS.

I switched to this dongle because it is smaller than the one from Canakit, which obviously works with PiPass. If this is the problem, then all I have to do is switch back to my old one when I want to use PiPass, since hostapd won't even start with it, assuming there's no way to configure hostapd to be compatible with it.

Matthew-Hsu, 2015-06-19:
The RTL8188CUS is known to be problematic with hostapd. There are ways to almost make it work, but it is unable to switch MAC addresses. I have heard that the RTL8188CUS can do this in Windows, but I assume the Linux drivers currently for it are shaky at best. It is possible that a driver may be released in the future that could correct this.

But yes, right now you would have to switch between WiFi dongles. 'Reset Network' will come in handy in this case, but there may be a good workaround for this if you just keep the Canakit WiFi plugged in and then you press 'Reset Network'. Once your RPi has been restarted, then you can plug your RTL8188CUS in. From there on, wlan0 should be reserved for your Canakit and wlan1 will be reserved for your RTL8188CUS. You should be able to freely switch without using 'Reset Network'.

closed 2015-06-19 by FamicomGuy