Matthew-Hsu/PiPass #38: PiPass not doing anything.

Buggzy1011, 2015-06-20:
When I set up PiPass using the automatic setup on a fresh install of rasbian running on a RBPi Model B  1st gen, it only will show the web interface. It will not save configs and will not start the service.

Matthew-Hsu, 2015-06-20:
Hey,

Just a few questions for you. Sometimes browser caching will display old values. Could you try manually refreshing the browser after you have saved your settings?

As for the services not starting up, are the logs generating anything for you in the dashboard?

The brand of your WiFi adapter and the chipset used would be good to know too.

If there is an issue with the automatic installer, I may be able to pinpoint what is wrong once these questions have been answered.

Thanks!

Buggzy1011, 2015-06-20:
I have tried to refresh the browser but will still not display the new values. I have also tried different browsers and devices but to the same problem.

The logs say no logging has been found.

I do not know the chipset but it is called a Wi-Pi. Here is a link to it: http://goo.gl/GnFDhG and http://goo.gl/YunAH6

Thanks for trying to help me.

Matthew-Hsu, 2015-06-21:
Hello again,

I am working with another user who is using the exact same RPi as you are. There seems to be differences that is causing issue. I am trying to figure out if it is a automatic installer issue or if I have to accommodate for a difference in PiPass itself.

Your WiFi chipset is Ralink based and it should work fine.

The other user I was working with reported that Raspberry Pi functionality in the dashboard does not work. Is this the same case for you? Could you run: 

    ->    chmod 755 -R /var/www 
    ->    chmod 755 -R /opt/PiPass

One last thing, could you copy and paste the output of this command: 

    ->    ps aux | grep apache

After chmod, you can see if the dashboard works as intended. I do have some users running a RPi Model B 512mb with the manual install just fine, but I am wondering if the RPi Model B 256mb is different enough to cause issue.

Thanks.

Buggzy1011, 2015-06-21:
The chmod ran fine, but the web interface is still doing the same thing. 

The logout output of the command is:

root      2373  0.0  2.7  24468 12376 ?        Ss   11:47   0:00 /usr/sbin/apache2 -k start
www-data  2396  0.0  1.8  24696  8424 ?        S    11:47   0:00 /usr/sbin/apache2 -k start
www-data  2399  0.0  1.3  24696  6168 ?        S    11:47   0:00 /usr/sbin/apache2 -k start
www-data  2404  0.0  1.3  24696  6168 ?        S    11:47   0:00 /usr/sbin/apache2 -k start
www-data  2405  0.0  1.8  24696  8416 ?        S    11:47   0:00 /usr/sbin/apache2 -k start
www-data  2406  0.0  1.8  24696  8416 ?        S    11:47   0:00 /usr/sbin/apache2 -k start
www-data  2734  0.0  1.3  24532  5932 ?        S    11:49   0:00 /usr/sbin/apache2 -k start
www-data  2742  0.0  1.2  24492  5524 ?        S    11:50   0:00 /usr/sbin/apache2 -k start
www-data  2743  0.0  1.2  24492  5524 ?        S    11:50   0:00 /usr/sbin/apache2 -k start
www-data  2744  0.0  1.2  24492  5524 ?        S    11:50   0:00 /usr/sbin/apache2 -k start
www-data  2745  0.0  1.2  24492  5524 ?        S    11:50   0:00 /usr/sbin/apache2 -k start
pi        2747  0.0  0.4   3548  1832 pts/0    S+   11:55   0:00 grep --color=auto apache

I am going to reinstall raspbian and try the manual install.

Also the model I have has 512MB of RAM as documented here: https://www.raspberrypi.org/products/model-b/

Buggzy1011, 2015-06-21:
I have just finished doing a manual install and PiPass works fine. Values save, the service can be controlled successfully, and I can streetpass people.

Matthew-Hsu, 2015-06-22:
Thanks for the feedback. I'll have to take another look at the installer and see what is happening.

The output you posted seem correct, so it should be something else.

Floflo81, 2015-07-15:
I had a similar issue. I got a 404 error when I tried to save the settings in the Web interface.
I fixed it myself: the problem was caused by the fact that the installer didn't install PHP 5, so PHP pages always gave a 404 error.
To fix that, I ran the following command:
apt-get install php5-common libapache2-mod-php5 php5-cli

Could you make it so the installer installs PHP 5?

Matthew-Hsu, 2015-07-16:
Thanks for the feedback.

I'm just wondering, are you using a Raspberry Pi Model B as well? So far the problem only seems to be happening on those running the original Raspberry Pi and those dependencies are actually automatically installed alongside PHP 5 and Apache on a RPi 2.

I have updated the installer to also include these dependencies. Hopefully, this will fix the problems in the future.

Thanks again!

Floflo81, 2015-07-16:
Yes I'm using a Raspberry Pi Model B+ with a standard Minibian install.

Matthew-Hsu, 2015-07-18:
These changes should work correctly and is related to another open issue at the moment. Did some re-work on the installer scripts, but the installer should work fine a standard Minibian install now.

Thanks for the feedback! I will be closing this issue now. If problems persist, feel free to reopen this ticket.

closed 2015-07-18 by Matthew-Hsu