Matthew-Hsu/PiPass #39: Wireless PiPass?

FamicomGuy, 2015-06-23:
Is there any way to configure PiPass to work wirelessly (no ethernet cable)? I tried using wicd-curses on a pre-installed image of PiPass, but all I was able to do with it is configure a static IP, which still relies on a wired connection. I can't get wicd to find any nearby networks.

It's not that I need this, I just want to know if it's possible. And I already tried the method involving two wifi dongles, which didn't work for me.

nagledb, 2015-06-23:
You said you already tried the method using two wifi dongles... what are you using now? Just one dongle with no wired connection? If so, that won't work. It needs two connections: one to get to the internet and the other to provide PiPass. One connection can't do both.

Technically, you can run it with just one and it'll cycle through networks. But since it can't reach the internet, you won't get StreetPasses.

FamicomGuy, 2015-06-23:
There's no good reason why PiPass should be wireless anyway.

closed 2015-06-23 by FamicomGuy