Matthew-Hsu/PiPass #42: 3DS cannot detect the wifi from pi.

birdy-chang, 2015-06-27:
Hi, I am using Asus WL-167G. I have installed PiPass on my raspberry pi model B and everything looks fine. I can see PiPass is running on the dashboard. MAC address is changing every 30 minutes. My computer and iPhone can detect the wifi form pi but not on my 3DS. I have also added MAC address of my computer and iPhone into /etc/hostapd/mac_accept and tried to connect to the wifi. But both of them showed "cannot join attwifi". Do you have any idea of this?

Matthew-Hsu, 2015-06-27:
Hello,

It's quite strange that your iPhone and computer can see the access point, but not your 3DS. Your RPi is broadcasting the access point, so it seems everything is running OK in that regard. If you disable Hostapd Security, do you still run into the issue of being able to not connect?

As for configuration, are you using the supplied configuration files from the PiPass manual install? There could be some networking issues that we are not seeing. Also, did you do the manual install on a pre-existing installation of Raspbian? I know hostapd did have issues like this before, so you may need to ensure that you are running the latest version of hostapd:

    ->    apt-get update
   ->     apt-get install hostapd

birdy-chang, 2015-06-28:
I had tried to disable Hostapd Security but still not work. I did the manual install on a fresh installed Raspbian and run hostapd with configuration file from the PiPass. I checked the hostapd version and it's the newest version.

Is there any way to check that this is a hardware or software issue? The USB dongle that I used is very old (about 10 years). I afraid that there is something broken inside.

Matthew-Hsu, 2015-06-28:
Hardware is a little bit tricky as it's easier to determine problems if you have a known working WiFi adapter to test with. In this case, you could try manually running hostapd via the command:

    ->    service hostapd start

Ultimately, PiPass just runs this command to get everything working. But, the logging system does detect that hostapd is running, so I do not know of a way to really determine this. When you manually installed, are you using the latest changes on the master branch? The logging system is a bit more intelligent and it should detect some common known issues with WiFi adapters. Though, if everything is running and PiPass is cycling, it is often the case of something with the WiFi adapter itself.

birdy-chang, 2015-06-28:
Yes, I installed the latest master branch. I have 2 wifi adapter, Asus wl167g v1(rt2571) and Edimax 7822uan(rtl8192cu), but both of them are not working perfectly with hostapd. That's really a bad luck :( . I will try to borrow a wifi adapter with a known working chipset. Thanks for your great help.

Matthew-Hsu, 2015-06-28:
Yes, that's a bit of bad luck there. Luckily some compatible chipsets, like the RT5370 can be found for a few dollars if need be.

I still find it strange that your computer and iPhone can see the network, but not your 3DS. Generally, the Ralink chipsets have been pretty good with people.

One last thing you could try is to remove the Ralink drivers and reinstall them:

    ->    apt-get install firmware-ralink

Once installed, I would use the PiPass Dashboard and use the 'Reset Network' feature under 'Raspberry Pi'.

I also tried Googling around for more information, but couldn't find anything regarding hostapd and RT2571.

Best of luck.

closed 2015-06-28 by Matthew-Hsu

birdy-chang, 2015-07-01:
I bought a Buffalo WLI-UC-GNM and everything work perfectly. I got homepass finally. lol

Matthew-Hsu, 2015-07-01:
Yay! Really glad everything is working out for you now. Hope you enjoy it!