Matthew-Hsu/PiPass #43: Internet Access Is Not Available

rickb47285, 2015-07-04:
I purchased the Canakit Starter Kit from your amazon.ca link. Installed the image onto the sd, I am able to access the dashboard. But when I press start my log just repeats

2015-07-04 07:42:46,486 - __main__ - WARNING - Internet access is not available. Trying to reconnect in one minute.

Is there something I missed. lsusb tells me that my adapter is a RT5370.

Matthew-Hsu, 2015-07-04:
Just to double check, do you have an Ethernet cable from your router plugged into your RPi? The setup should have an Ethernet cable plugged into your RPi with the WiFi adapter plugged in as well. The WiFi adapter will not be providing your RPi with Internet.

If you have this setup already and still have problems, could you press 'Reset Network' in the PiPass Dashboard and try again? If problems are still happening, could you try pinging a website like www.google.com in the terminal of your RPi to see if your RPi actually has Internet access?

Thanks!

rickb47285, 2015-07-04:
Yep, that's how I am set up. I have hit Reset Network many times with no luck. Google ping responds with no problem. Thanks for the reply.

Matthew-Hsu, 2015-07-04:
That's quite strange. Are you able to ping 8.8.8.8 and 8.8.4.4?

These are Google ran servers that I use for a network connectivity test and I am wondering if there is an issue with pinging those servers.

Which method of installation are you using as well?

rickb47285, 2015-07-04:
I used the image download. I cannot ping those dns servers. And this is for sure the reason this doesn't work. My network is blocking those severs so that my Google Chromecast can access American Netflix, Hulu etc..

Is there any way I can change those servers in the network connectivity test? Thanks again.

Matthew-Hsu, 2015-07-04:
Ah, OK. This makes more sense now. I can definitely address this issue and issue a fix that addresses these cases in the future.

In the meantime, if you are comfortable with editing a file, could you do this:

    ->    nano /opt/PiPass/piPass.py

Press CTRL + W and find the words: while True

You should be in a section that looks like this:

# Determines the network connectivity status of PiPass.
def isNetworkConnected():
    global isDisconnected

    while True:

You'll want to change True to False. So it should look like this in the end:

# Determines the network connectivity status of PiPass.
def isNetworkConnected():
    global isDisconnected

    while False:

This will disable all network connectivity tests.

Matthew-Hsu, 2015-07-04:
If it is easier, you can just replace the file in /opt/PiPass/piPass.py with this one here:

https://drive.google.com/open?id=0B8bbfqFbkJvVRkhYYzRKcDdMWWs

For the time being, I wouldn't use PiPass Update, but there haven't been any updates since the 1.6 release, so you won't be missing any new features.

rickb47285, 2015-07-04:
I updated that function and everything is working properly now. Thanks again for the quick help.

Matthew-Hsu, 2015-07-04:
No worries and sorry for inconvenience. I do want to address this issue better and I was wondering with your current setup, do you also block Open DNS servers as well? For example, 208.67.222.222.

I probably should do a dual approach and use Google and Open DNS, instead of an entire Google approach.

Thanks!

nagledb, 2015-07-04:
Maybe use docs.google.com for the ping? (If you can't reach that, you can't load the spreadsheet so it's broken regardless.)

rickb47285, 2015-07-04:
@Matthew-Hsu I do not block OpenDNS, just Google. Thanks again.

Matthew-Hsu, 2015-07-04:
@nagledb Yes, I thought about that too. I used IP addresses instead just in case if the DNS server being used was not working correctly. At this point, you would still have Internet access, but just have a non-working DNS server to debug with your router. But, I'm not sure if I need to be this accurate since if the DNS server was not working correctly, PiPass really wouldn't work too.

@rickb47285 I've added OpenDNS to the network connectivity test, so it should work in your case. If it doesn't feel free to let me know. If you use PiPass Update, you'll receive these changes and any future versions of PiPass will not require you to make the manual code changes each time.

closed 2015-07-04 by Matthew-Hsu