Matthew-Hsu/PiPass #44: PiPass not passing

commguy, 2015-07-17:
For the last week, I've seen no streetpasses on any of the 6 3DS systems that regularly use my relay. I've checked configuration, logs and settings on all devices. The system seems to have just "stopped". 

I'm using a Pi rev B
Asustek RT2573 wifi card

Syslog on Pi shows when a device associates, no errors in the PiPass logs.

I can connect a 3DS to the accesspoint (Pi) but it will not access the internet unless I put google DNS directly in the 3DS config.

I can however connect my phone to the accesspoint (Pi) and browse the internet.

The Pi itself does resolve DNS. 

I would say it looks like the Pi isn't passing DNS but if it is, it's only an issue passing DNS for 3DS consoles. 

Since the latest image says it's only for the Pi2 I've reflashed my card with 1.52, which use to work fine but now does not. I've also updated from 1.52 to 1.6 and I've been unsuccessful in getting this to work. I tried switching to different spreadsheets today and was able to get it working for about 20 min and then nothing again.

I'm a little stumped here. I can't imagine an issue with the Pi itself causing this, and the wifi card has been working like a champ since I set this up. 

Any insight would be appreciated.

commguy, 2015-07-18:
Ok. So..... This is a little embarrassing. Aparrently my Active Directory Server was on the blink. I noticed the internet seemed a little slow and I was having trouble with DNS resolution on my own network. Since my DC is the first DNS on my network I rebooted it and I could browse the internet on my test console again. Apparently this happened when my UPS dropped my servers last weekend. My DC server died and I resurrected it with a spare chassis, it was authenticating users on my network but apparently wasnt fully operational. I'll be quietly working on that now.

Sorry for wasting a post.

closed 2015-07-18 by commguy

Matthew-Hsu, 2015-07-18:
Glad that you were able to figure out what was going on with your network! Sometimes this stuff is a bit tricky, so don't worry about "wasting a post".

I hope you enjoy PiPass!