Matthew-Hsu/PiPass #46: [Question] I updated Pipass after not having done so for a while.  I see this Hostpad Security?

Corruptinglyneedful, 2015-07-20:
What is hostpad security and should I enable it?

Hey, I see you finally got the ability to add 3DS Mac's to the Dashboard, that's great!

Matthew-Hsu, 2015-07-21:
Hey,

I think you were last running 1.3? In 1.4, I introduced the enabling and disabling of security. If you turn off hostapd security, any 3DS in range will be able to use your WiFi and effectively get StreetPasses. I also think in 1.4, adding 3DS MAC addresses were also added.

All the other additions from since have mostly been quality of service with the exception of the progress bar in 1.6.

Corruptinglyneedful, 2015-07-21:
What do you mean in range?  I thought they had to be authorized through the mac list?

Matthew-Hsu, 2015-07-22:
That's what Hostapd Security controls. If Hostapd Security is checked, only 3DS' that are on the 3DS MAC whitelist will only be authorized to get StreetPasses. Otherwise, if Hostapd Security is unchecked, any 3DS will be able to get StreetPasses.

PiPass will broadcast a WiFi (eg., attwifi), so by range, I mean any device that can connect to that WiFi.

Corruptinglyneedful, 2015-07-22:
Well that's not good.  That apparently is not enabled.  Time to enable that!

closed 2015-07-23 by Matthew-Hsu