Matthew-Hsu/PiPass #48: Internet disconnecting

lucky644, 2015-07-28:
Hello,

Running the PiPass with the image for the Raspberry Pi 2.

Runs great, except that it seems very unstable.  It frequently loses the internet connection (from the ethernet plugged directly into my router)

This is all new to me, and I don't even know where to begin looking for how/why this is happening.

All I can do right now is provide you with the only logs I found.

2015-07-28 16:31:50,263 - __main__ - INFO - PiPass is now running.
2015-07-28 16:32:14,869 - __main__ - INFO - Spoofing as 02:53:50:4F:4F:63 on attwifi ( Extended Mac ) for 30 minute(s).
2015-07-28 16:33:07,327 - __main__ - WARNING - Internet access is not available. Trying to reconnect in one minute.
2015-07-28 16:34:18,036 - __main__ - INFO - Spoofing as 02:53:50:4F:4F:35 on attwifi ( Extended Mac ) for 30 minute(s).
2015-07-28 16:36:44,346 - __main__ - WARNING - Internet access is not available. Trying to reconnect in one minute.
2015-07-28 16:37:55,087 - __main__ - INFO - Spoofing as 6E:53:50:4F:4F:90 on attwifi ( Extended Mac ) for 30 minute(s).
2015-07-28 16:39:44,216 - __main__ - WARNING - Internet access is not available. Trying to reconnect in one minute.
2015-07-28 16:40:54,975 - __main__ - INFO - Spoofing as 02:53:50:4F:4F:08 on attwifi ( Extended Mac ) for 30 minute(s).
2015-07-28 16:42:44,083 - __main__ - WARNING - Internet access is not available. Trying to reconnect in one minute.
2015-07-28 16:43:54,829 - __main__ - INFO - Spoofing as 02:53:50:4F:4F:21 on attwifi ( Extended Mac ) for 30 minute(s).
2015-07-28 16:45:36,858 - __main__ - WARNING - Internet access is not available. Trying to reconnect in one minute.
2015-07-28 16:46:47,612 - __main__ - INFO - Spoofing as 6E:53:50:4F:4F:11 on attwifi ( Extended Mac ) for 30 minute(s).

Any ideas how I can troubleshoot this?

Matthew-Hsu, 2015-07-29:
PiPass pings OpenDNS and falls back to Google DNS to test an outside connection.

If you are familiar with Linux, could you ping 208.67.222.222 and see if the connection is stable? How is your Internet on other devices? Is the Internet stable on your RPi?

The only time I have heard of a similar issue is that if you have blacklisted those DNS servers on your router, but since PiPass moves onto another Nintendo Zone, the connection seems to get interrupted. Could you also list which WiFi adapter you are using?

lucky644, 2015-07-29:
Terribly sorry for not posting my wifi adapter.

It is from the Canakit, 

http://www.canakit.com/raspberry-pi-wifi.html

Chipset Ralink 5370

I am able to ping that ip from my desktop without issues, 25ms no dropped packets.  I have a -t ping going on while I'm trying to get the Pi to work, and when it is dropping the internet connection, there is no drop from my desktop.

I have not actually tried logging into the Pi directly, when I bought it I simply installed your image to it.

I'll see if I can find a spare keyboard and go log in.

lucky644, 2015-07-29:
Forgot to mention, all my other devices have a rock solid connection to my internet, I have not had any issues at all.

lucky644, 2015-07-29:
Ok, I logged into the Pi directly, and I've been pinging that IP, and it isn't dropping.

I tried to SSH in to make this easier, but the connection is refused (port 22 using putty) is it disabled by default?

lucky644, 2015-07-29:
Well, what seems to be happening is that wlan0 is leaving promiscuous mode, shuts down the eth0, and cycles again.  No direct error message seen, and pinging continues without a hitch (it hangs but it never says it dropped)

Hopefully these pics help..

[image]
[image]
[image]

Matthew-Hsu, 2015-07-29:
Port 22 should be open. I almost exclusively only use SSH when using my RPi.

This should work and I am wondering if this is related. Another guess I have is that it could be possible that your router may have some type of traffic management that does not like outgoing periodic pings?

If you look at issue #43, this will disable network connectivity tests, but if there is something going on, you may not get full StreetPasses if the connection is unstable. However, if your router just does not like periodic pings, I can look into enabling/disabling this test.

lucky644, 2015-07-29:
Very odd.  I have a D-Link DIR-868L, and I'll take a look but I don't think there is anything specific in there that would block this....

I could try DMZing the Pi?  Or would that be a bad idea?

Matthew-Hsu, 2015-07-29:
If the pinging is still stable even when it has that hiccup, the network test could be falsely failing. I'll have to take a closer look when I get home, but you may want to see if you can get SSH working and see if that fixes this issue.

If not, you could temporarily follow issue #43 and turn off the test as you have shown the connection is still stable after the hiccup.

lucky644, 2015-07-29:
Ok, I followed the directions and changed True to False, it has been stable ever since.

So I guess that fixed it, thanks!

I would like to note, however, since i saw something similar mentioned on that other thread, I use unblock.us so I get USA netflix etc.

I wonder if this is related?

lucky644, 2015-07-29:
Guess I spoke too soon, after over 30 min of working great, the whole thing crashed, bunch of error messages on the screen, mostly about USB stuff, I restarted it and I haven't been able to log into the web ui since.

I also setup a ping to the Pi itself, the ping WILDLY varies, anywhere from 2ms all the way to 207ms

It no longer keeps a connection to the network itself.

lucky644, 2015-07-29:
Update:

I wiped the install, redid the image, rebooted my router, checked all connections and booted it back up.

I put the Pi into DMZ in the router.

Now, my pings to the Pi are <1ms HOWEVER, the disconnecting is still happening.

I still cannot SSH into the Pi via putty port 22 either.

lucky644, 2015-07-29:
2015-07-28 19:50:05,208 - __main__ - INFO - PiPass is now running.
2015-07-28 19:50:17,595 - __main__ - INFO - Spoofing as 00:15:C6:29:5F:92 on attwifi ( McDonald's on Route 29 in Chillicothe, Illinois ) for 15 minute(s).
2015-07-28 20:05:29,431 - __main__ - INFO - Spoofing as FC:0A:81:58:AE:E2 on KPN ( Albert Heijn XL Zaandam, NH, NL ) for 15 minute(s).
2015-07-28 20:20:40,431 - __main__ - INFO - Spoofing as 6E:53:50:4F:4F:0F on attwifi ( Extended Mac ) for 15 minute(s).
2015-07-28 20:35:52,235 - __main__ - INFO - Spoofing as 40:53:50:4F:4F:17 on attwifi ( Extended Mac ) for 15 minute(s).
2015-07-28 20:51:02,423 - __main__ - INFO - Spoofing as 4E:53:50:4F:4F:6A on attwifi ( Extended Mac ) for 15 minute(s).
2015-07-28 21:06:12,273 - __main__ - INFO - Spoofing as 6E:53:50:4F:4F:1D on attwifi ( Extended Mac ) for 15 minute(s).
2015-07-28 21:38:47,106 - __main__ - WARNING - A possible incompatible hostapd WiFi chipset has been detected.
2015-07-28 21:38:47,108 - __main__ - ERROR - Unable to change the MAC address of the WiFi adapter.
2015-07-28 21:38:47,111 - __main__ - INFO - PiPass has been shutdown with an error.

lucky644, 2015-07-30:
I guess I have to give up, I am installing SpillPass, I prefer yours 10x but it simply isn't working for some reason.

Matthew-Hsu, 2015-07-31:
Hey,

Sorry, I have been busy lately. A lot Linux Homepass based solutions are built around Hostapd. The logging system is generally used to report errors with hardware to help those who may have a WiFi issue.

Looking at the logs, PiPass does run for a bit, but for some reason your WiFi adapter did not change MAC addresses. I believe PiPass is the only solution that checks for these issues, but if Spillpass works for you, that is great. Though, if your WiFi adapter is not changing MAC address and if other solutions do not check for that, then you may still have problems, but it is just hiding.

If the WiFi adapter is dropping connection after numerous pings and is not changing MACs, there could be an issue with the WiFi adapter. Considering you are using the same kit as me and many others and are reporting this issue for the first time, there could be something with the adapter.

But, it's really hard to tell without having another WiFi adapter to test against.

I will be heading out for a trip, so I won't be able to provide much support. piPass.py is commented quite well and if you are OK with commenting blocks of code, you can comment out the MAC address verification part. If you do, PiPass will look like it is running, but if this will just ignore the WiFi issues that you are having.

I hope this helps.

lucky644, 2015-07-31:
I'm not sure what the issue is, SpillPass has been running for 3 hours now without any issues (yet) and I am getting streetpasses.

I'll keep a eye on it for 24 hours and see if it is still stable.  If it is, it doesn't really seem like it would be a hardware issue.

Matthew-Hsu, 2015-07-31:
Yes, it is interesting. The only thing I am worried about is that it could be a partial hardware issue. Partial meaning that your WiFi adapter is only sometimes not actually switching MAC addresses.

When you are running SpillPass and unplug your Ethernet cable from your RPi, does it continue running? If it does, then for that moment when you don't have Internet access then you won't be able to receive StreetPasses, so the problem is just being hidden.

Assuming you are not getting the disconnections after moving your RPi to a DMZ, you'd have to double check the MAC address on your WiFi adapter to make sure it is actually spoofing the MAC address of the Nintendo Zone. Even in cases where the MAC addresses are not switching, you could still get StreetPasses in these cases, but not from the correct zone.

The logging could possibly be too strict in this scenario and won't allow for any recovery if the expected outcome is not being read. If you were to download PiPass 1.4, it would be interesting to see what the behaviour is. That version does not do any internet connectivity tests or verify the MAC address switch. You would have to overwrite /opt/PiPass/ and /var/www/ with the ones included and make sure the correct permissions are set.

Though, not sure if that would be worth it in this scenario. If I can reproduce the issue in my environment, I'll make sure to keep you updated.

lucky644, 2015-08-01:
It seems Spillpass is not stable either.

Any ideas on how I can start troubleshooting this...?  Could this be a faulty wifi adaptor from Canakit?

Matthew-Hsu, 2015-08-02:
Are you still getting the dropped connections via ping? If your network reconfiguration fixed that, I can only assume from the logs that the failed MAC switch is caused by a possibly faulty WiFi adapter. The MAC switch should always occur and not sometimes.

I have an automatic installer that you can use if for some reason the PiPass or SpillPass pre-made images cause some fault in your environment. You can run the installer from a clean install of Raspbian and see if the problem still occurs.

If the problem persists, I would probably save your router settings to a file and try the pre-made image again with factory default router settings to rule out some type of network configuration.

If all if these fail, I would suspect a faulty WiFi adapter.

lucky644, 2015-08-03:
I wiped the spillpass install and installed it again shortly after my last post, it has been stable and working ever since.  I have been monitoring the logs and mac address changes via ssh, and both of my 3ds units have been getting streetpasses and no pings have been dropped.

I didn't do anything different this time except I did run sudo apt-get update and sudo apt-get upgrade.

The Pi is still in DMZ mode.

I'd still prefer to run PiPass, but now that I finally got it stable I don't want to mess with it anymore haha.

Matthew-Hsu, 2015-08-07:
Yeah, I am not too sure. I'll keep a look out in case I experience this issue or if someone else reports the same thing.

closed 2015-08-07 by Matthew-Hsu