Matthew-Hsu/PiPass #49: dhcpcd failure on reboot

bsweeti, 2015-08-06:
Hi there.  Using a raspberry pi (1) model B with TP-LINK TL-WN7200ND,  after both automatic and manual installs, dhcpcd fails to start on reboot "because /etc/network/interfaces... Failed!"

I'm definitely not an experienced linux user, any help would be greatly appreciated.

Matthew-Hsu, 2015-08-07:
A quick double check on your setup. Do you have an Ethernet cable plugged from your router to your RPi? If so, double check that you only have one WiFi adapter connected to your RPi as well. In case you have had more than one WiFi adapter connected to your RPi at any given time, could you access the PiPass dashboard and click 'Raspberry Pi' -> 'Reset Network'.

Hopefully this will fix your issue. If not, could you paste in the output of 'ifconfig' here so that I can take a look?

Thanks.

bsweeti, 2015-08-07:
Yes the ethernet is directly connected from the router and only the one wifi adapter.  This is also a fresh install as well.

Once Pipass is installed (automatically), the RPi network is down and I can no long SSH in... so here's a pic of the ifconfig:

[image]

Thanks for the help.

bsweeti, 2015-08-09:
Just did a fresh install, here is my current /etc/network/interfaces:

auto lo
iface lo inet loopback

auto eth0
allow-hotplug eth0
iface eth0 inet manual

auto wlan0
allow-hotplug wlan0
iface wlan0 inet manual
wpa-conf /etc/wpa_supplicant/wpa_supplicant.conf

auto wlan1
allow-hotplug wlan1
iface wlan1 inet manual
wpa-conf /etc/wpa_supplicant/wpa_supplicant.conf

There is a wlan1 listed... not sure why...  Here is iwconfig:

wlan0     IEEE 802.11bgn  ESSID:off/any
          Mode:Managed  Access Point: Not-Associated   Tx-Power=20 dBm
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:off

lo        no wireless extensions.

eth0      no wireless extensions.

Let me know if you need any more info.

bsweeti, 2015-08-09:
And here is lsusb:

Bus 001 Device 002: ID 0424:9512 Standard Microsystems Corp.
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp.
Bus 001 Device 004: ID 0e8f:0022 GreenAsia Inc.
Bus 001 Device 005: ID 148f:3070 Ralink Technology, Corp. RT2870/RT3070 Wireless Adapter

closed 2024-08-31 by Matthew-Hsu