Matthew-Hsu/PiPass #52: Dashboard not showing up

Schiggy69, 2015-11-15:
Hi,

I'm using a B model and a TN-WN725n v2.0 (rtl 8188eu).  My installation went smooth with the 
automatic installer on Raspian and I nearly double checked everything. The problem is all I get when I type in the pi's ip is the default raspian/apache page and not your dashboard. I'm out of things I can try to make PiPass work, and I hope you can help me somehow.

robplease, 2015-12-02:
I have this exact problem too. I can't get Pipass to work at all. 

What I noticed is that the installation puts the front end html files into the \var\www\ folder, but when you go to your IP address in a browser, it looks for html files located in the \var\www\html\ folder only. When I move all the front end files into the \var\www\html\ folder I can see the front end in my browser, but nothing works at all because it is in a different folder than expected.

Novax95, 2015-12-05:
I had this issue using the automated installer. After messing with it, I found that the Apche server was saying that /var/www/html was the document root, while PiPass wanted to live in /var/www.

I simply edited two files:
- /etc/apache2/sites-available/000default.conf
     - Change "DocumentRoot /var/www/html/" to "DocumentRoot /var/www/"

- /etc/apache2/sites-available/default-ssl.conf
     - Change "DocumentRoot /var/www/html/" to "DocumentRoot /var/www/"

After editing the two above mentioned files, you should be able to see the dashboard, assuming none of the html files or assets folder were moved from /var/www.

Schiggy69, 2015-12-06:
Thanks, that helped. But now I'm running into a new problem. 

2015-12-06 11:41:13,978 - __main__ - INFO - PiPass is now running.
2015-12-06 11:41:22,659 - __main__ - WARNING - A possible WiFi driver issue has been detected.
2015-12-06 11:41:22,662 - __main__ - ERROR - Unable to start hostapd.
2015-12-06 11:41:22,670 - __main__ - INFO - PiPass has been shutdown with an error.

I manually installed the  rtl8188eu but I'm tired of this driver. I also had issues with Spillpass because of this thing.

Novax95, 2015-12-06:
If I've understood the collective Internet correctly, that driver is for a Realtek chipset, which I am not currently using. 

However, I did locate this issue that advocates that the RLT8188 driver is known for having issues with Hostapd: https://github.com/Matthew-Hsu/PiPass/issues/37

If it helps, I'm using the dongle provided in this CanaKit (Used Matthew's referral link):
http://www.amazon.com/gp/product/B008XVAVAW/ref=as_li_tl?ie=UTF8&camp=1789&creative=390957&creativeASIN=B008XVAVAW&linkCode=as2&tag=matthew08b-20&linkId=IN557BBG3PJEAU7M

I didn't see a referral link listed by Matthew, but this is the wireless dongle by itself:
http://www.amazon.com/CanaKit-Raspberry-Wireless-Adapter-Dongle/dp/B00GFAN498/ref=sr_1_14?s=electronics&ie=UTF8&qid=1449420716&sr=1-14

As a warning though, I too am getting driver error messages with the above mentioned wireless dongle whenever PiPass tries to automatically spoof the next MAC address or I use the "Advance" option in PiPass. I did manage to find a work around, so for all intents and purposes, it's effectively working 100%. It's just doing it in a slightly different way. 

Here's the link to my issue which explains my issue\workaround in case you decide to purchase that dongle, or if you run into similar problems down the road: https://github.com/Matthew-Hsu/PiPass/issues/53

closed 2024-08-31 by Matthew-Hsu