Matthew-Hsu/PiPass #54: Updating settings will not work

robplease, 2015-12-05:
When I try to update the settings on the PiPass dashboard I am brought to 

<My IP Address>/assets/php/settings.php

and it just shows the php file. 

I made sure all the files were in the correct location during the installation. I used the manual installation method.  I can't tell what there error is here, if it can't find a file or something. I am using Raspian

Corruptinglyneedful, 2015-12-13:
ctrl+F5

There is an issue with pages caching.

robplease, 2015-12-13:
It seems to be behaving weridly for me. That's what it does in Chrome, but when I try it in IE11 it takes me to a page that says the Dashboard path isn't correct/not found. Which is weird because I made sure that the file permissions were correct and that it is installed in the correct /var/www/ folder.

timHT, 2016-01-22:
I did manual installation (Raspbian) and apache has the default site in /var/www/html and not /var/www.  I had to change /etc/apache2/sites-available/000-default.conf to use "DocumentRoot /var/www" before things worked.

closed 2024-08-31 by Matthew-Hsu