Matthew-Hsu/PiPass #55: Hostapd Issue

kaeyonles, 2015-12-24:
I can't get this to work as I'm just getting the below error every time. Below links are to the components that I'm using and I've tried with the security both on and off, Any suggestions on what I can try next (please note that I don't have any experience with linux or programming)? 

2015-12-23 21:40:58,986 - __main__ - INFO - PiPass is now running.
2015-12-23 21:41:20,579 - __main__ - WARNING - A possible WiFi driver issue has been detected.
2015-12-23 21:41:20,580 - __main__ - ERROR - Unable to start hostapd.
2015-12-23 21:41:20,583 - __main__ - INFO - PiPass has been shutdown with an error.

WiFI dongle - http://www.ebay.com.au/itm/231505879138?_trksid=p2060353.m2749.l2649&ssPageName=STRK%3AMEBIDX%3AIT 
Raspberry Pi Kit - http://www.ebay.com.au/itm/252142186992?_trksid=p2060353.m2749.l2649&ssPageName=STRK%3AMEBIDX%3AIT

kaeyonles, 2016-01-06:
Just wondering if this has been looked into at all?

Nodeswitch, 2016-02-28:
I'm having the same issue as well with the above mentioned adapter, along with the two below on a Raspberry Pi B 2 on the custom image you provide. 

```
Ralink Technology, Corp. RT5370 Wireless Adapter
Realtek Semiconductor Corp. RTL8188CUS 802.11n WLAN Adapter
```

Billtopia, 2017-02-09:
try my solution on issue https://github.com/Matthew-Hsu/PiPass/issues/80 here

it works with --> Ralink Technology, Corp. RT5370 Wireless Adapter

closed 2024-08-31 by Matthew-Hsu