Matthew-Hsu/PiPass #58: Cannot get PiPass to work

oobedoob, 2016-02-07:
I have a Raspberry Pi 2 and cannot get PiPass to work.

I've tried the manual and automatic installs and whenever I go to the IP address I get the Apache2 Debian Default Page.

I've tried the PiPass image and I can access the page, but it keeps erroring out with "unable to start hostapd"

I've used a Ralink RT5370 and a Edimax Realtek RTL8188CUS wireless adapter.

oobedoob, 2016-02-07:
Got it working. I was using Jessie, tried Wheezy and it started working.

sokka11, 2016-02-19:
I have the same problem. Where did you get Wheezy from? I can't find it anywhere.

oobedoob, 2016-02-19:
There was a link on the Raspberry Pi's site for older version downloads, but it looks like they took it down. I did find a site that has a list of mirrors that do have Wheezy available for download. https://www.raspbian.org/RaspbianMirrors

nagledb, 2016-03-29:
Rasbian Wheezy is no longer available on that link, but you can download the last version of it here: https://downloads.raspberrypi.org/raspbian/images/raspbian-2015-05-07/

Billtopia, 2017-02-09:
you have to go to the /etc/apache2/sites-available folder and edit the 000-default.conf file

change 
```DocumentRoot /var/www/html``` to ```DocumentRoot /var/www``` and it should work

closed 2024-08-31 by Matthew-Hsu