Matthew-Hsu/PiPass #61: Disable Mac Filtering??

xabertheblade, 2016-02-13:
Bad practice, I know, but. . .Is there anyway to disable the MAC address filtering, or set it to a more permissive mode? I have quite a few friends that would like to partake in the street pass smorgasbord, however, I just don't want to deal with inputting so many DS MACS. I don't know how inadvisable this is, from a Nintendo-gets-pissy standpoint, but it would certainly simplify matters for me.

kabili207, 2016-04-18:
In the settings menu, uncheck the `Hostapd Security` option.

>I don't know how inadvisable this is, from a Nintendo-gets-pissy standpoint, but it would certainly simplify matters for me.

Nintendo wouldn't care in the least bit. MAC filtering is used to prevent people from using your otherwise open access point for whatever they want.

xabertheblade, 2016-04-18:
That seems much easier than I was expecting ;p Thank you very much for your reply :)

closed 2016-04-18 by xabertheblade