Matthew-Hsu/PiPass #64: Ralink RT5370 Hostapd Issues

snarfie, 2016-03-08:
```
2016-03-08 04:11:48,786 - __main__ - INFO - PiPass is now running.
2016-03-08 04:11:56,523 - __main__ - INFO - Spoofing as 40:53:50:4F:4F:92 on attwifi ( Extended Mac ) for 5 minute(s).
2016-03-08 04:16:58,101 - __main__ - WARNING - A possible invalid MAC address has been detected: 6E:53:50:4F:4F:87.
2016-03-08 04:16:58,102 - __main__ - ERROR - Unable to start hostapd.
2016-03-08 04:16:58,104 - __main__ - INFO - PiPass is moving onto the next Nintendo Zone.
2016-03-08 04:16:58,717 - __main__ - WARNING - A possible incompatible hostapd WiFi chipset has been detected.
2016-03-08 04:16:58,718 - __main__ - ERROR - Unable to change the MAC address of the WiFi adapter.
2016-03-08 04:16:58,721 - __main__ - INFO - PiPass has been shutdown with an error.
```

This is the event log. I am using Raspberry Pi 2 and a Ralink RT5370 chipset. I have verified the chipset using -sudo lsusb and have updated the drivers. This is a fresh set of raspbian Jesse and the newest PiPass. Are there any suggestions or comments on how to correct this?

DebuRanger, 2016-03-09:
lsusb output
Bus 001 Device 004: ID 148f:5370 Ralink Technology, Corp. RT5370 Wireless Adapter

Debian Version 8.0 Jessie

I can't even get a web gui to work.  I just get the standard apache page up.
I'm also on a Rasp B+

Please let me know how I can assist.
I did use your autoinstaller.

DebuRanger, 2016-03-09:
OK, got the gui up.
The installer put them in 
/var/www
They need to go into 
/var/www/html

DebuRanger, 2016-03-09:
OK.  New update.  Sorry
Had to modify 
/etc/apache2/sites-available/000-default.conf

Changed this line
DocumentRoot /var/www/html
to
DocumentRoot /var/www

Working great now.

DebuRanger, 2016-03-09:
OK, Now its stuck at Zionsville.  It won't push another Mac Address nor will it let me advance manually.

DebuRanger, 2016-03-09:
OK.  Figured that one out.
There is no init script to start the pyPass.py code.
Do you have one somewhere?

DebuRanger, 2016-03-09:
Here is the init script I borrowed from the internet to make this work.
You need to put it in /etc/init.d/piPass

 #! /bin/sh
# /etc/init.d/example

case "$1" in
  start)
    echo "Starting PiPass"
    # run application you want to start
    python /opt/PiPass/piPass.py &
    ;;
  stop)
    echo "Stopping PiPass"
    # kill application you want to stop
    killall python
    ;;
  *)
    echo "Usage: /etc/init.d/example{start|stop}"
    exit 1
    ;;
esac

exit 0

DebuRanger, 2016-03-09:
Make sure you add the runlevels for 0, 5, and 6.
Otherwise the script won't work.

It seems to be fine now. 
Nice software.

DebuRanger, 2016-03-10:
Forgot to add LSB stuff.

#! /bin/sh

### BEGIN INIT INFO
# Provides:        piPass
# Required-Start:  $network $remote_fs $syslog
# Required-Stop:   $network $remote_fs $syslog
# Default-Start:   2 3 4 5
# Default-Stop: 
# Short-Description: Start NTP piPass
### END INIT INFO

# /etc/init.d/example
 
case "$1" in
  start)
    echo "Starting PiPass"
    # run application you want to start
    python /opt/PiPass/piPass.py &
    ;;
  stop)
    echo "Stopping PiPass"
    # kill application you want to stop
    killall python
    ;;
  *)
    echo "Usage: /etc/init.d/example{start|stop}"
    exit 1
    ;;
esac
 
exit 0

closed 2024-08-31 by Matthew-Hsu