Matthew-Hsu/PiPass #65: Pi can use WiFi adapter, but PiPass won't start?

mrllamatastic, 2016-03-10:
I've just freshly installed using the image file on a brand new Pi 2 I got today. (Complete noob)

Remoting into the PiPass Dashboard  fine, and entered the MAC for my DS, but no matter how many times I try clicking Refresh or Start, the PiPass always says "PiPass is not running". I'm (ignorantly) assuming that this is something to do with the WiFi dongle I've plugged into the Pi as I hadn't previously attempted to connect to the internet via WiFi.

Doing an iwlist scan on the command line shows me all the networks in range, so surely that means the Pi is successfully using the WiFi dongle? Why can't PiPass, then?

DebuRanger, 2016-03-10:
See Issue number 64.  I think I had to run piPass manually in /opt.

mrllamatastic, 2016-03-11:
Thanks for the reply, DebuRanger

I took a look at the issue you mentioned, but sorry to say I can make very little sense of it (like I said, I'm a noob).

From what I can gather, you were unable to get the GUI. I can get it fine, it's just the "Start PiPass" doesn't seem to be doing anything.

Loreinator, 2016-03-11:
Can you post the logs from the gui page? That may help us identify the
cause.

mrllamatastic, 2016-03-12:
`2016-03-12 07:56:42,794 - __main__ - INFO - PiPass is now running.
2016-03-12 07:57:03,113 - __main__ - WARNING - A possible WiFi driver issue has been detected.
2016-03-12 07:57:03,114 - __main__ - ERROR - Unable to start hostapd.
2016-03-12 07:57:03,117 - __main__ - INFO - PiPass has been shutdown with an error.`

This is what the log's saying

mrllamatastic, 2016-03-12:
^ I'm guessing this means that while the WiFi adapter may be fine for the Pi itself, it's not compatible with PiPass?

mrllamatastic, 2016-03-17:
**UPDATE**: I ordered the WiFi adapter that was recommended in the ReadMe rather than using the one that came bundled with my Pi, and no everything's working fine. Just posting this so that people getting similar errors know what to do!

closed 2024-08-31 by Matthew-Hsu