Matthew-Hsu/PiPass #66: WPA2 authentication for attwifi?

RShadowhand, 2016-03-14:
Using RPi3 and latest version of PiPass. I want to add some sort of password to my attwifi because lots of people connect to it and use the internet.

I edited hostapd.conf to add some password, but my 3DS couldn't find it then, and starting PiPass re-writes the file every time.

Loreinator, 2016-03-14:
Use the mac filter, they won't be able to connect if you have it limited by mac address to your 3DS only.

RShadowhand, 2016-03-14:
Mac filter didn't work. I could still connect to it using my iPhone.

zennoux, 2016-03-15:
DId you get PiPass working with the built-in wifi on RPi 3?

RShadowhand, 2016-03-15:
Yeah. The image didn't work, it'd hang on black screen. So I downloaded the latest Raspbian version, and installed PiPass with the shell script (the installer one), and it just worked like that.

closed 2024-08-31 by Matthew-Hsu