Matthew-Hsu/PiPass #68: No config page when connecting via web browser

Blunderbuss, 2016-03-18:
Only the apache test page shows up

JosiahDub, 2016-03-19:
I also get this. I installed via install_pipass.sh. I tried to run piPass.py but received this error:

Traceback (most recent call last):
  File "./piPass.py", line 32, in <module>
    handler = logging.handlers.RotatingFileHandler(LOG_FILENAME, maxBytes=1048576, backupCount=0)
  File "/usr/lib/python2.7/logging/handlers.py", line 117, in __init__
    BaseRotatingHandler.__init__(self, filename, mode, encoding, delay)
  File "/usr/lib/python2.7/logging/handlers.py", line 64, in __init__
    logging.FileHandler.__init__(self, filename, mode, encoding, delay)
  File "/usr/lib/python2.7/logging/__init__.py", line 911, in __init__
    StreamHandler.__init__(self, self._open())
  File "/usr/lib/python2.7/logging/__init__.py", line 936, in _open
    stream = open(self.baseFilename, self.mode)
IOError: [Errno 13] Permission denied: '/opt/PiPass/logs/piPass.log'

nagledb, 2016-03-31:
The "apache test page" issue is discussed with a possible solution in #69 for anyone else who comes along.

@JosiahDub you shouldn't need to run piPass.py directly from the command line, hopefully the fix I reference above will let you use the web dashboard. If not, then I suspect the issue is you were running piPass.py as a normal user. Try running it this way instead:

`sudo python /opt/PiPass/piPass.py`

closed 2024-08-31 by Matthew-Hsu