Matthew-Hsu/PiPass #69: Auto-installer does not work for RPI3

Blunderbuss, 2016-03-19:
It's not updated yet for the RPI3, and unfortunately it seems the author may have stopped supporting it?

pSykAwtiX, 2016-03-29:
Just picked up a RPI3 for this very project. Looks like I should of bought an RPI2 instead... I hope he hasn't stopped supporting it, but I can't find much evidence to the contrary.

Blunderbuss, 2016-03-29:
Yeah, it looks like he has :(

nagledb, 2016-03-29:
Can you describe what isn't working? Do you see any error messages during the installation process? Depending on what messages you're seeing, maybe I or someone else can make some suggestions that might help you to get it working.

Also, what linux distro and version are you trying to install under? That is likely to have a large impact on whether it succeeds or not and will be important to know for troubleshooting.

nagledb, 2016-03-29:
#58 describes similar problems and they got it working with Wheezy. I recommend trying Wheezy to see if it works for you. Looks like you can download the last Wheezy release here: https://downloads.raspberrypi.org/raspbian/images/raspbian-2015-05-07/

I have no idea if it'll work with RPI3 but perhaps it's worth trying.

pSykAwtiX, 2016-03-29:
I was using the Raspbian Lite distro from the RPi site and the installer itself appears to work great. I only watched the script up to the point where it was downloading WIFI drivers but when I came back, a reboot had taken place. But, when I put the IP of the Pi in a web browser, I just get the **default Apache site**. I tried an additional reboot but I still had the same issue.

I went over the manual install instructions and tried a few thing. I assumed the dependencies downloaded just fine because I didn't see any errors at that point. I'm a bit skeptical if the WIFI drivers step was necessary or may of caused issues. Probably not but I know that the RPI3 has built in WiFI. I did the "sudo chmod -R 755" for both "/opt/PiPass/" and "/var/www/". I also verified that the changes were made to sudoers. Finally, I went through etc and var to make sure that the PiPass files were downloaded and everything appeared to be there. 

At that point, I was planning on following through the script step by step when I had more time to see if I could get an error. Though, it could be worth wile to try another distro.

pSykAwtiX, 2016-03-31:
> I have no idea if it'll work with RPI3 but perhaps it's worth trying.

I'm just tried the latest Raspbian Wheezy but couldn't get it to work with the RPi3. Just hangs at the rainbow square. After lots of research, it looks like Wheezy is not compatible with the RPi3.

nagledb, 2016-03-31:
I believe I found the problem: it looks like Jessie changed the web root from /var/www/ to /var/www/html. In the install script (install_pipass.sh), you want to change DASHBOARD_PATH (around line 27) so that the line reads:

`DASHBOARD_PATH="/var/www/html/"`

Then it should install and work properly.

nagledb, 2016-03-31:
Oops, just discovered another step required. Once it's installed, you'll also have to change a setting in PiPass. In the Dashboard, select "PiPass", then "Settings". You need to change "Dashboard Path" (should be last in the list) to "/var/www/html/".

pSykAwtiX, 2016-04-01:
You did it! **The "DASHBOARD_PATH="/var/www/html/" edit within the install script was all that was needed!**

I was able to make it dashboard right after it finished. I did notice the reference to "/var/www/" in the settings but I left it alone at that moment. After I added my n3ds mac address, streetpasses started rolling in! I'm pretty excited to be finally participating in that feature. 

A few minutes later, I lost access to the dash board completely. That's when I got your second message about the necessity of editing the dashboard location within the dashboard itself, which i should of done already. I had to rerun the script to gain access to the dashboard once again to make that change. It's now been over a half hour and the dashboard has not disappeared, so I think I'm good to go there. 

One other thing that was a bit strange was that the dash board status would not reflect the fact that the pipass service was running and it did not display the time remaining until next zone, but it was all working anyways due to the fact that I was indeed receiving streetpasses. After a half hour, even that started working suddenly. 

Looks like after about 6 months of messing around with homepass options, I've finally have a solution that works! **I cannot thank you enough, nagledb!**

ediazfer, 2016-04-30:
For me it only started working when, after a few minor fixes like the ones told above, I removed systemd from the system.  Before there were issues with hostapd.

urbanhusky, 2016-09-27:
**Alternative fix for the "only the default apache website shows" issue:**

Edit the apache virtual host config (`/etc/apache2/sites-enabled/000-default`) and change the config so that it points to `/var/www` instead of `/var/www/html`

That worked on my PI3 and I did not touch the install script.

syelle, 2018-03-11:
I did the following to the [Dropbox version of the install script](https://www.dropbox.com/sh/pe2v63t476pmg9n/AADfRQVSMloUKkOhmEkGcjdwa?dl=0&preview=install_pipass.sh) get PiPass working on the Raspberry Pi 3:

Line 27: Change the DASHBOARD_PATH variable in "install_pipass.sh" to "/var/www/html/":

    `DASHBOARD_PATH="/var/www/html"`

Lines 44-47: Replace the lines that install PHP. They refer to older, invalid package names.

Old:

    `$SUDO apt-get install php5 -y > /dev/null 2>&1`
    `$SUDO apt-get install php5-common -y > /dev/null 2>&1`
    `$SUDO apt-get install php5-cli -y > /dev/null 2>&1`
    `$SUDO apt-get install libapache2-mod-php5 -y > /dev/null 2>&1`

New:

    `$SUDO apt-get install php -y > /dev/null 2>&1`
    `$SUDO apt-get install php-cli -y > /dev/null 2>&1`
    `$SUDO apt-get install libapache2-mod-php -y > /dev/null 2>&1`

Run install_pipass.sh.

closed 2024-08-31 by Matthew-Hsu