Matthew-Hsu/PiPass #75: Pi3 issue

Percolateur2000, 2016-09-15:
Hi

When i start pipass i have this on the logs

2016-09-15 12:56:21,948 - __main__ - INFO - PiPass is now running.
2016-09-15 12:56:26,555 - __main__ - INFO - Spoofing as 02:53:50:4F:4F:0D on attwifi ( Extended Mac ) for 5 minute(s).
2016-09-15 13:01:32,686 - __main__ - WARNING - A possible invalid MAC address has been detected: 6E:53:50:4F:4F:6F.
2016-09-15 13:01:32,687 - __main__ - ERROR - Unable to start hostapd.
2016-09-15 13:01:32,688 - __main__ - INFO - PiPass is moving onto the next Nintendo Zone.
2016-09-15 13:01:33,290 - __main__ - WARNING - A possible incompatible hostapd WiFi chipset has been detected.
2016-09-15 13:01:33,291 - __main__ - ERROR - Unable to change the MAC address of the WiFi adapter.
2016-09-15 13:01:33,294 - __main__ - INFO - PiPass has been shutdown with an error.

i did `apt-get update` but nothing changed

Pipass settings :
 
StreetPass Cycle Time (Minutes) : 5
Hostapd Security : unchecked
Hostapd Driver : nl80211
Dashboard Path : /var/www/html/

(had to change according to another issue with pi3 and automatic installation)

I did the automatic installation wihtout any problem. The onboard wifi chipset is BCM43143. A

ttached file is my current interfaces.

Any idea ? Thank you !!

[interfaces.txt](https://github.com/Matthew-Hsu/PiPass/files/474446/interfaces.txt)

Overload86, 2017-06-18:
I have the same problem on my Pi3. It changes the MAC once or twice and then says it can't anymore. Used your automatic installation script on a retropie image with raspbian on lxde desktop. Which by the way is putting the whole config website into /var/www and installs httpd, but forgets to edit the httpd config file. But once I figured out where that was saved it was easy to edit.
Website works, one or two mac changes work and then I get the error in the log.

person65278, 2017-07-28:
this is a copy and paste of my answer to basically the same issue, well my fix fixes both issues
This quick fix is for a rpi3
I fixed this by changing and adding some code to pypass.py. I basically made it so when it checked if it had internet, if not it would take eth0 down then back up. And also when hostapd goes belly up I have it set to try again instead of just shutting down. Here is my modified pyPass.py. put it into /opt/PiPass/pyPass.py or wherever your original pyPass.py file is.......

[piPass.zip](https://github.com/Matthew-Hsu/PiPass/files/1181622/piPass.zip)

closed 2024-08-31 by Matthew-Hsu