Matthew-Hsu/PiPass #79: /var/www used on raspbian jessie rather than /var/www/html

cbozic, 2017-01-09:
First, thank you.  Second, I can confirm that pi pass installs pretty much as you describe following the "Automatic Installer" with one modification.  The default directory for apache is not /var/www instead of /var/www/html.  So, I had to "sudo cp -R /var/www/html /var/www" after the install.

FYI, I'm also testing pi pass with the "third-party" Ubuntu-MATE OS.  It seems to have the same apache directory issue mentioned above but I can't say I've got it working past that.  It looks like an issue with the network bridge that I haven't had enough time to learn about.

I've done all of this with a Raspberry PI 3, BTW.

Thanks again for the hard work on this.

ourovoros, 2017-01-09:
Hi cbozic,

As far as I know, the hostapd package which need to be create bridge has been broken somehow since Ubuntu 14. I believe latest Ubuntu-MATE based on Ubuntu 16 so. It's also broken. I think it's not PiPass issue but related Ubuntu issue.

https://bugs.launchpad.net/ubuntu/+source/wpa/+bug/1289047
http://askubuntu.com/questions/472794/hostapd-error-nl80211-could-not-configure-driver-mode

Maverynthia, 2017-03-13:
I had the same problem on Raspbian Jessie. Copying the files to /html/ worked.

closed 2024-08-31 by Matthew-Hsu