Matthew-Hsu/PiPass #80: not starting hostapd nor writing new network_configuration file 100% of the time

Billtopia, 2017-02-09:
I have found that piPass.py doesn't always manage to start hostapd 100% of the time and it also doesn't write out the hostapd.conf file 100% of the time either.

It was causing errors saying my usb wifi driver was not compatible and also that the hardware could not change the MAC so it was incompatible even after piPass had been running for some time.

I modified piPass.py to attempt to start hostapd several times (always seems to restart on the first retry) and the first time the MAC doesn't change I have it rewrite the hostapd.conf file and restart hostapd again and now I get no errors running for hours on end. It would die out on the 3rd/4th point change before.

not sure if there is a better way to do this or not but...

```python

#!/usr/bin/python

#### Imported Libraries ####

import os
import io
import time
import json
import urllib
import signal
import os.path
import logging
import datetime
import subprocess
import logging.handlers

from random import shuffle

#### Logging Support ####

LOG_FILENAME = '/opt/PiPass/logs/piPass.log'

# We only want to keep the log file of the current execution of PiPass.
if os.path.isfile(LOG_FILENAME):
    subprocess.call('sudo rm ' + LOG_FILENAME, stdout=open(os.devnull, 'w'), stderr=open(os.devnull, 'w'), shell=True)

# Set up a specific logger with the desired output level.
logger = logging.getLogger(__name__)
logger.setLevel(logging.INFO)

# Add the log message handler to the logger.
handler = logging.handlers.RotatingFileHandler(LOG_FILENAME, maxBytes=1048576, backupCount=0)

# Create a logging format.
formatter = logging.Formatter('%(asctime)s - %(name)s - %(levelname)s - %(message)s')
handler.setFormatter(formatter)

logger.addHandler(handler)

#### Support Functions ####

# Read and load the path to where the PiPass Dashboard is installed.
def loadDashboard():
    # Read the settings.
    try:
        with open('/opt/PiPass/config/pipass_dashboard.json', 'r') as f:
            pipass_dashboard = json.load(f)
    except IOError:
        logger.error('Unable to read the file: /opt/PiPass/config/pipass_dashboard.json.')
        updateStatus('Not Available', 'Not Available', 'PiPass is not running')
        logger.info('PiPass has been shutdown with an error.')
        exit(1)

    # Controls the location to where the PiPass Dashboard is installed.
    global DASHBOARD

    # Path to the JSON file where PiPass will write to for the PiPass Dashboard to display connection information.
    global DASHBOARD_INFO

    # Path to the JSON file where PiPass will write to for the PiPass Dashboard to calculate the time until the next Nintendo Zone.
[... 489 lines cut ...]
                f.write(unicode(json.dumps(results['feed']['entry'][currentZoneIndex], ensure_ascii=False)))
                f.write(unicode(']'))
        except IOError:
            logger.error('Unable to write the file: ' + DASHBOARD_INFO + '.')
            updateStatus('Not Available', 'Not Available', 'PiPass is not running')
            logger.info('PiPass has been shutdown with an error.')
            exit(1)

        currentZoneIndex = currentZoneIndex + 1

        # Receiving SIGUSR1 or SIGUSR2 will interrupt the time.sleep call. The loop allows
        # for sleep to be resumed up to the current cycle setting.
        start = (datetime.datetime.utcnow() - datetime.datetime(1970, 1, 1)).total_seconds()
        while (datetime.datetime.utcnow() - datetime.datetime(1970, 1, 1)).total_seconds() - start < STREETPASS_CYCLE_SECONDS:
            # Verify network connection. If network connectivity is lost, we want to halt StreetPass cycling and attempt to reconnect.
            # Once reconnected, move to the next Nintendo Zone in the list.
            if isNetworkConnected() and isDisconnected:
                break

            time.sleep(5)

logger.info('PiPass has been shutdown successfully.')

exit(0)
```

ceo75, 2017-02-09:
I have the same issue with my Pi3 running on the internal wifi adapter.  If this fixes the issue, well done!!  Might not have a chance until this weekend to try it though.

Billtopia, 2017-02-10:
so... I may have found a faster way without all the added code.

1 new line and 1 small change. I am testing it now to see if the changes work (if they don't the other changes will catch the problem and log it)

the change has been running for some time on mine and no logged errors being caught by my previous additions.

I stop hostapd before writing the hostapd.conf file
write the file
then start (not restart) hostapd

this also has the bonus of shaving like 8 seconds off the time because there is a sleep(8) in the init file for hostapd when you do a restart.

```python
# Write the current zone information to NETWORK_CONFIGURATION.
        subprocess.call('sudo service hostapd stop', stdout=open(os.devnull, 'w'), 
        try:
            fo = open(NETWORK_CONFIGURATION, "w")
        except IOError:
            logger.error('Unable to write the file: ' + NETWORK_CONFIGURATION + '.')
            updateStatus('Not Available', 'Not Available', 'PiPass is not running')
            logger.info('PiPass has been shutdown with an error.')
            exit(1)

        conf = "interface=wlan0\nbridge=br0\ndriver=" + HOSTAPD_DRIVER + "\nssid=" + zoneValues[0] + "\nbssid=" + zoneValues[1] + "\nhw_mode=g\nchannel=6\nauth_algs=1\nwpa=0\nmacaddr_acl=" + HOSTAPD_SECURITY + "\naccept_mac_file=/etc/hostapd/mac_accept\nwmm_enabled=0\nignore_broadcast_ssid=0"
        fo.write(conf)
        fo.close()
        
        # Restart hostapd to ensure NETWORK_CONFIGURATION is used. Restarting hostapd will also ensure that it is running if it is currently off.
        # subprocess.call() will wait for the service command to finish before moving on.
        subprocess.call('sudo service hostapd start', stdout=open(os.devnull, 'w'), stderr=open(os.devnull, 'w'), shell=True)
```

Billtopia, 2017-02-10:
So, after 20+ MAC address changes it hasn't triggered the first set of code changes I put in. I think this has solved it.

ceo75, 2017-04-06:
Hi @Billtopia ,

I apologize for taking so long to get your changes tested on my pi3, life got in the way.  :)

After loading up the modified pipass.py changes everything appears to be working great!

Well done!

Billtopia, 2017-04-07:
Glad it is working for you.

closed 2024-08-31 by Matthew-Hsu