Matthew-Hsu/PiPass #82: User and Password infinite loop error.

NGamerCole, 2017-02-24:
Basically this is what happens. I go to enter the User, I type it in just fine and press enter. Then when it gets on the password, I can't type anything at all, the only thing I can type in first is the enter key which will put it to the next line down and it's taking it as the whole password. And it just loops this over and over. 

I formatted the microSD card and followed the instructions, so I'm guessing there is some particular error?

NGamerCole, 2017-03-08:
Anything?

hackerx31337, 2017-03-11:
If i understand you correctly your problem is that you don't have a problem, just a misunderstanding. I'm guessing this is your first time dealing with Linux. The password prompt is supposed to stay blank no matter what you type. It is a security feature. Not only can somebody watching not tell what the password you are entering is, they can't even see how long it is.

person65278, 2017-07-28:
What hacker said..... Type the password hit enter you wont see anything you typed and you will be logged in as long as you typed the correct password.

Corruptinglyneedful, 2017-08-09:
There a way to disable this?

Confused me too, never used to do that.

closed 2024-08-31 by Matthew-Hsu