Matthew-Hsu/PiPass #83: WiFi driver issue

purrfectlyme, 2017-03-11:
Hi,

I have this problem,

2017-03-10 23:04:33,723 - __main__ - INFO - PiPass is now running.
2017-03-10 23:04:57,080 - __main__ - WARNING - A possible WiFi driver issue has been detected.
2017-03-10 23:04:57,083 - __main__ - ERROR - Unable to start hostapd.
2017-03-10 23:04:57,090 - __main__ - INFO - PiPass has been shutdown with an error.

Using Ralink RT5370 WiFi Dongle
Raspberry Pi Model B+ 512MB

Any ideas?

Billtopia, 2017-03-11:
You could try and replace your pipass.py file with this one and see if it helps.

https://github.com/Billtopia/PiPass/blob/patch-1/opt/PiPass/piPass.py

Maverynthia, 2017-03-15:
I had the same problem, this fixed it so :3

Billtopia, 2017-03-15:
I am glad it helped... I am going to make a few more tweaks to it as it seems to have a few rare instances where it can still bug out... (I have seen it do it like 5 times in 1 week @5 mins per zone) but it is 99.9% reliable right now.

closed 2024-08-31 by Matthew-Hsu