Matthew-Hsu/PiPass #84: PiPass kills network connection

Ferkner, 2017-03-18:
I manually installed PiPass on a Raspberry Pi 3 running Raspbian Jessie. After going through the setup and making the tweaks to get the dashboard page to display in the browser, I started PiPass. Once I did, it killed the wifi connection on my Pi. wlan0 is associated with attwifi, Configured 169.254.64.119. That is the IP I have to put in the browser to get the dashboard. Anything else does not work. There is no connection to the internet at all, so PiPass cannot read the Google spreadsheet database. So anytime I start the service, it stops on an error.

A couple years ago I was never able to get SpillPassPi to work either unless I installed it with a pre-made image. If I do that with this, can I install other things on the Pi as well? I wanted this Pi 3 to do double duty as a StreetPass relay as well as a Kodi media center. Can I still do that if I install PiPass via pre-made image? Or do I need to fix this issue?

I am new to the whole linux thing, so you'll have to explain things to me as you would explain something to an idiot :).

Billtopia, 2017-03-18:
So, is it connected to your lan with an ethernet cable as well? or are you running 2 seperate wifi adapters (well the built in one and a separate one) Does the rest of the PiPass web interface work as well (can you get to it or is it the apache default setup page?

Ferkner, 2017-03-18:
It was only connected with the built-in wifi. I suppose it needs a second connection as well then? Makes sense now that I remember that SpillPassPi needed an Ethernet connection as well.

I wasn't lying about the idiot part in my comment! I'll give that a try and report back.

Ferkner, 2017-03-18:
No luck plugging the Ethernet in either. br0 has an IP of 192.168.2.188, and eth0 had an IP of 192.168.2.186. I changed eth0 so that it was 192.168.2.188 as well, but that did nothing. I have no clue what to do apart from starting from scratch or going with the pipass image.

Billtopia, 2017-03-18:
ok   while on the pi open the weg browser and nagigate to 127.0.0.1 and do you see the piPass page or the apache setup page

Billtopia, 2017-03-18:
and leave the network devices set to auto configure.

Ferkner, 2017-03-19:
I see the PiPass page. I moved all the PiPass stuff from /var/www/ to /var/www/html/ because I knew there was an issue with the apache page coming up instead of the pipass dashboard. I'm thinking I broke something and it might be a pain to unbreak.

Billtopia, 2017-03-19:
did you change the did you change the path on the pipass dashboard config tab as well?(PiPass -->settings)  after that what does the log say when you click start

Ferkner, 2017-03-19:
I did. I changed it to /var/www/html/. I did this all last night on the initial setup. The log says there is an error accessing or reading the google docs spreadsheet, and that pipass was stopped. The entire Raspberry pi has no internet access.

Billtopia, 2017-03-19:
I would do this https://www.leowkahman.com/2016/03/19/disable-ipv6-raspberry-raspbian/

and in /etc/network the interfaces file it should look something like this:
```
auto lo
iface lo inet loopback

auto br0
iface br0 inet dhcp
    bridge-ports eth0 wlan0

auto wlan0
allow-hotplug wlan0
iface wlan0 inet dhcp
```

Ferkner, 2017-03-19:
No dice.

I think I'll start over. Maybe I'll give the pipass image a try, or try the manual reinstall again since I want this Pi to run kodi as well.

Billtopia, 2017-03-19:
I did the install script on Jessie w/ Pixel and only had the change the root for the web server(changed the 000 config) and the wifi driver for hostapd on the pi2. I had tried the image first but couldn't get it to work

Ferkner, 2017-03-21:
This program simply doesn't work. I tried reinstalling it for the third time, following the manual install instructions, and it just doesn't work.

Billtopia, 2017-03-22:
is it the same issue where you loose internet on the pi? you can try this http://askubuntu.com/questions/79000/what-is-the-equivalent-terminal-command-to-ipconfig-all to make sure the eth0 is set up right and did wired work right before it got PiPassed

person65278, 2017-07-28:
This quick fix is for a rpi3
I fixed this by changing and adding some code to pypass.py. I basically made it so when it checked if it had internet, if not it would take eth0 down then back up. And also when hostapd goes belly up I have it set to try again instead of just shutting down. Here is my modified pyPass.py. put it into /opt/PiPass/pyPass.py  or wherever your original pyPass.py file is.......

[piPass.zip](https://github.com/Matthew-Hsu/PiPass/files/1181612/piPass.zip)

closed 2024-08-31 by Matthew-Hsu