Matthew-Hsu/PiPass #87: Pi pass Pi 3

Vspin99, 2017-06-09:
I have had this working but now it boots and then goes to the log on screen.
Goes into starting 2 times then to log in and sits
Have reflashed the SD card and same result.
Help please

person65278, 2017-07-27:
This is a month old but I was able to get this working on a rpi3 from a fresh install.

Vspin99, 2017-07-28:
image would be great
thank you

Vspin99, 2017-08-17:
any luck on the image?

person65278, 2017-08-18:
I had forgot about it. but here it is. Username is pi  password is
raspberry. image should be plug and play for an rpi 3. It is made from
https://github.com/Matthew-Hsu/PiPass  I made it with a fresh raspbian
install then made a cpl of the dependencies from source....... Changed some
code so it would retry instead of crashing on some errors that by what i
saw were unavoidable at random times. sorry for the size I fixed the errors
but I have never done much when it comes to making disk images so ​
 pipassrpi3.zip
<https://drive.google.com/file/d/0B_-ruduXYrW1RGZPdGlxeTlQeFk/view?usp=drive_web>
​someone who knows how to shrink them is more then welcome to to a crack at
it and take it down in size.

DuelUtility, 2017-11-12:
@person65278 Any chance you still have that image?

closed 2024-08-31 by Matthew-Hsu