Matthew-Hsu/PiPass #91: Only getting 2 or 3 people per streetpass relay?

Corruptinglyneedful, 2017-08-14:
What is up with this, surely the community for this is not small so what's going on here.

Why am I not getting 10 per relay and how do I fix it?  

 Sometimes I don't get any passes, either which doesn't make sense to me given the sheer amount of relays on the list.

ViridianDragoon, 2017-08-14:
Each NZone can only store data for three users. Once a new user that isn't already storing their data at that NZone is within range the oldest data gets replaced. That's why it's said you shouldn't hit the same zone in at least 8 hours to allow other users to have their data stored. If your data is already stored into one of those three spots then you won't receive data from that NZone.

Corruptinglyneedful, 2017-08-14:
Since when?  It used to be 10 or 5, not sure.

I did know about the 8 hr limit but our list has hundreds of zones.  I have it set to shuffle the list.

Shuffles every 30 min

ViridianDragoon, 2017-08-14:
Unless Nintendo changed it recently it's always been three per spot. You might be getting confused with the limit on how many passes Mii Plaza can hold.

closed 2024-08-31 by Matthew-Hsu