Matthew-Hsu/PiPass #93: Manual install on Raspberry Pi B3 raspbian stretch

lemgandi, 2017-12-31:
Worked out of the box.  I used PHP 7 rather than 5 and thought that'd be a trouble, but everything Just Worked.  I'm using a Raspberry Pi B 3 running Raspbian 4.9.41-v7 (Stretch Lite) on a 4 gb SD card.  I installed Stretch Lite on the card with dd(1) from my linux box.  I was able to get onto my wifi network by editing /etc/wpa-supplicant/wpa-supplicant.conf .  Once I had her talking to the net,  I installed the following packages with apt-get(8):

apache2     
bridge-utils
hostapd
p7zip-full
php7.0

The other packages in the README.md list were either already installed in Stretch or already in the php7.0 metapackage.

I then unpacked PiPass-master and copied its contents to the appropriate directories in root ( /etc and /opt ).  When I rebooted, my 3ds green light came on and the web interface was on port 8080 just as expected.  Great job man.

lemgandi, 2017-12-31:
I found that the assets/php/pipass_logs.php script was not finding /opt/PipPass/logs/piPass.log.  The symptom was that it displayed the "no logs found..." error message.  I fiddled with this for a whiles before adopting the Brute Force & Ignorance method of making a hardlink to /opt/PiPass/logs/piPass.log in assets/logs/piPass.log and changing the location referenced in pipass_logs.php.  I'm sure there's a better way to do this in /etc/php/7.0/php.ini or /etc/apache2/apache2.conf (or both).

dh47, 2018-02-16:
I updated the auto-installer script to install PHP7 modules instead.  Download it if you wish.
[install_pipass.zip](https://github.com/Matthew-Hsu/PiPass/files/1730529/install_pipass.zip)

closed 2024-08-31 by Matthew-Hsu